TomKing062/CVE-2022-38694_unlock_bootloader #2: Hisense a5 pro CC root failed

zhinu, 2023-06-26:
Hisense A5PRO cc root failed. 

Here Arch linux, 

```
./spd_dump exec_addr 0x3f48 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec read_part teecfg 0 1M teecfg.bin read_part trustos 0 6M tos.bin read_part sml 0 1M sml.bin erase_part splloader reset
libusb_control_transfer ok
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
input >
```

After inputing the code from the readme, I end up with an input > asking me to input some code... 

I can help debug, should you want it and I also have another Hisense device with a unisoc t610 the hisense q5, that I would like to root. So if you want my help, let me know :)

zhinu, 2023-06-26:
With --verbose 1
```

./spd_dump --wait 5000 --verbose 1 exec_addr 0x5500 fdl uboot-mod.bin 0x9efffe00 exec read_part teecfg 0
 1M teecfg.bin read_part trustos 0 6M tos.bin read_part sml 0 1M sml.bin erase_part splloader reset
Waiting for connection (5000s)
libusb_control_transfer ok
send: unknown message
recv: type = 0x81, size = 6
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
send: type = 0x00, size = 0
recv: type = 0x80, size = 0
CMD_CONNECT bootrom
input >
```

TomKing062, 2023-06-27:
use  spreadtrum_flash-main_exec instead of interactive

ferlanero, 2023-06-27:
> use spreadtrum_flash-main_exec instead of interactive

I'm using Android 10 with software version L1732.6.17.01.00 in Hisense A5PRO cc This root method apply with this software version? And where to use `spreadtrum_flash-main_exec`? I can't find where it is

zhinu, 2023-06-27:
> > use spreadtrum_flash-main_exec instead of interactive
> 
> I'm using Android 10 with software version L1732.6.17.01.00 in Hisense A5PRO cc This root method apply with this software version? And where to use `spreadtrum_flash-main_exec`? I can't find where it is

git clone the repo, and build within the folder

ferlanero, 2023-06-27:
> git clone the repo, and build within the folder

Is it necessary to prepare the phone in any way before applying root?

zhinu, 2023-06-27:
> use spreadtrum_flash-main_exec instead of interactive

now I have this issue

 ```
libusb_control_transfer ok
send: unknown message
recv: type = 0x81, size = 6
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
send: type = 0x00, size = 0
recv: type = 0x80, size = 0
CMD_CONNECT bootrom
current exec_addr is 0x5500
loadfile("uboot-mod.bin") failed
```

ferlanero, 2023-06-27:
> now I have this issue
>
> ```
> libusb_control_transfer ok
> send: unknown message
> recv: type = 0x81, size = 6
> CHECK_BAUD bootrom
> BSL_REP_VER: "SPRD3\0"
> send: type = 0x00, size = 0
> recv: type = 0x80, size [...]

After reaching this step, does the phone still start?

TomKing062, 2023-06-27:
> 
1. i tested a t610 today, the offset is wrong, fdl1 can't be executed without verify
2. spd_dump can't use `exec_addr` now, rename method like `spd_dump_3f48` works

will fix later

TomKing062, 2023-06-27:
fdl1 execute fixed https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/commit/048596daadbec75156e23d591c24b2bcdc4c0441

zhinu, 2023-06-28:
Nice keep up the good work. Any way to support you, donation?

zackyjsd, 2023-06-30:
感谢大佬的工作和分享关注这个洞很久了，终于等到有可用的exp，顺便提一下，这个手机的root有几个人在论坛悬赏几百美金，如果成功root，大佬绝对是第一功臣。
https://forum.xda-developers.com/t/hisense-a5-pro-cc-how-to-root-unlock-bootloader.4109021/page-8

TomKing062, 2023-07-07:
fdl1 execute fixed https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/commit/0482afcb3177b896c2816cb80d129a5099d64c3f
download tool from releases

TomKing062, 2023-07-07:
now fdl1 can be executed without verify
though i am still not sure if fdl1 from SmileX(t610) can be used to unlock a5p/a5pcc (ddr init is slightly different)

TomKing062, 2023-07-07:
about spl method, not written yet
device will not unlock if ddr init in fdl1 failed , if this happen , modify spl and ddr init will use exactly value from manufacturer, i will see if fdl1 can work first

TomKing062, 2023-07-07:
smilex fdl1 not work, a5p/a5pcc still need compile fdl1 manually
this takes about 1 day

zhinu, 2023-07-07:
> smilex fdl1 not work, a5p/a5pcc still need compile fdl1 manually this takes at least 1 day

Good work, could you document how to compile fdl1 for other t610 devices? (I have a hisense q5) :)

TomKing062, 2023-07-08:
DONE

closed 2023-07-08 by TomKing062

Xinon232, 2023-07-08:
thanks :)
Unfortunately I can't install the root... the executable only says: Waiting for device....

However adb does recognize my phone, spd_dump.exe doesn't... any solution advice?

TomKing062, 2023-07-23:
q5 added

zhinu, 2023-07-24:
AMAZING