TomKing062/CVE-2022-38694_unlock_bootloader #4: Would be this applicable to unisoc SC9863A if I have root already?

Skorpion96, 2023-07-04:
I have the ZTE Blade A5 2019, it is unisoc SC9863A. I was able to root it by flashing a prerooted gsi, now I would like to unlock bootloader. The thing now is: if I flash the edited images with dd will this method work for my device? I mean I don't need to hack download mode as I can flash partitions with dd, that said with root would this method be applicable to my device? I can provide all the firmware images that you may need if necessary.

TomKing062, 2023-07-04:
If device had hash key fused, secureboot will stop edited images being executed. 
if device is fused, why user can flash gsi
if not fused, unlock bootloader should be possible (dd partition is not a good idea)

Skorpion96, 2023-07-04:
I'll explain better, the thing on my device was that I flashed an engeneering firmware that has dm-verity disabled, and seems that flash with research download tool isn't checked. That's why I was able to flash and boot a gsi prerooted. But I checked with an app and bootloader is still locked. I tried to flash some custom recovery that some dev (a friend of mine) made but I'm unable to boot anything custom. That's why I was hoping in this exploit. I mean I could use dd or if dd isn't good i can flash from download mode. So the question is would be this exploit (in particular editing the partitions to make bl unlocked) compatible with my device?

TomKing062, 2023-07-05:
compatible but useless

> secureboot will stop edited images being executed

closed 2023-07-05 by TomKing062

TomKing062, 2023-11-05:
9863a added now, I didn't have enough understanding before, the process you mentioned is feasible, with root we can unlock bootloader, the point is modify uboot(uboot_a) partition while keep original uboot_bak(uboot_b)

Skorpion96, 2023-11-05:
Oh, awesome, thank you, I'll provide you uboot and uboot_bak. If you need anything else tell me

TomKing062, 2023-11-05:
dd from engineering firmware, need 4 partitions: uboot uboot_bak mmcblk0boot0 mmcblk0boot1

Skorpion96, 2023-11-05:
Here are files attached
[uboot, uboot_bak, mmcblk0boot0, mmcblk0boot1.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/13259816/uboot.uboot_bak.mmcblk0boot0.mmcblk0boot1.zip)

Skorpion96, 2023-11-05:
there is the first issue:C:\Program Files (x86)\ADB and Fastboot++>adb root
* daemon not running; starting now at tcp:5037
* daemon started successfully
restarting adbd as root

C:\Program Files (x86)\ADB and Fastboot++>adb shell
s9863a1h10:/ #
C:\Program Files (x86)\ADB and Fastboot++>adb shell
s9863a1h10:/ # dd if=/storage/emulated/0/uboot_unlock.bin of=/dev/block/by-name/uboot
1026+1 records in
1026+1 records out
525412 bytes (513 K) copied, 0.510903 s, 0.9 M/s
s9863a1h10:/ # dd if=/dev/zero of=/dev/block/mmcblk0boot0
dd: /dev/block/mmcblk0boot0: write error: Operation not permitted
1+0 records in
0+0 records out
0 bytes (0 B) copied, 0.000258 s, 0 B/s
1|s9863a1h10:/ # dd if=/dev/zero of=/dev/block/mmcblk0boot1
dd: /dev/block/mmcblk0boot1: write error: Operation not permitted
1+0 records in
0+0 records out
0 bytes (0 B) copied, 0.000270 s, 0 B/s
1|s9863a1h10:/ #
C:\Program Files (x86)\ADB and Fastboot++>adb shell
s9863a1h10:/ # dd if=/dev/zero of=/dev/block/by-name/mmcblk0boot0
dd: /dev/block/by-name/mmcblk0boot0: write error: Operation not permitted
1+0 records in
0+0 records out
0 bytes (0 B) copied, 0.000425 s, 0 B/s
1|s9863a1h10:/ #

As you can see mmcblk0boot0 and mmcblk0boot1 are read only, what do i do now?

Skorpion96, 2023-11-05:
Maybe i can flash an empty mmcblk0boot0 and mmcblk0boot1? About the boudrate i guess the exe needs to be modified

TomKing062, 2023-11-05:
CHECK_BAUD is check connection, connect failed due to erase failed
try `adb reboot autodloader`
or AT command `echo 'AT+SPREF=\"AUTODLOADER\"\\r' >/dev/stty_lte0` (not sure about name stty_lte0, choose the least one)

Skorpion96, 2023-11-05:
well "adb reboot autoloader" seems to be working, the device is on black screen, but "spd_dump exec_addr 0x4ee8 fdl spl_unlock.bin 0x5000" give previous error:
spd_dump exec_addr 0x4ee8 fdl spl_unlock.bin 0x5000
current exec_addr is 0x4ee8
Waiting for connection (30s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD4:AutoD\0"
CMD_CONNECT bootrom
SEND FDL1
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FAIL
CHECK_BAUD FAIL
CHECK_BAUD FAIL
CHECK_BAUD FAIL
wrong command or wrong mode detected, reboot your phone by pressing POWER and VOL_UP for 7-10 seconds.

C:\Users\admin\Downloads>spd_dump exec_addr 0x4ee8 fdl spl_unlock.bin 0x5000
current exec_addr is 0x4ee8
Waiting for connection (30s)
wrong command or wrong mode detected, reboot your phone by pressing POWER and VOL_UP for 7-10 seconds.

Skorpion96, 2023-11-05:
Ok i tried to split the unlock command in two
[image]
it seems to work but the phone remains on autodloader without going into recovery, and if i reboot the phone bootloader is still locked.

TomKing062, 2023-11-06:
SPRD4:AutoD\0 is in uboot, SPRD3\0 is in bootrom
i thought you didn't have FDLs
`spd_dump fdl fdl1-sign.bin 0x5000 fdl fdl2-sign.bin 0x9efffe00 exec erase_part splloader reset`

Skorpion96, 2023-11-06:
ok that's what happened
[image]
it didn't went to recovery and key attestation app says bootloader is locked

Skorpion96, 2023-11-06:
Ok i tried to split the "spd_dump fdl fdl1-sign.bin 0x5000 fdl fdl2-sign.bin 0x9efffe00 exec erase_part splloader reset" in two parts and it did something, now if i go on the key attestation app it gives some error regarding some certificate, i consider this good but if i then go check bootloader status from terminal it results locked
C:\Users\Admin
[image]
[image]
\Documents\ZTE Blade A5 2019\Various\CVE-2022-38694_Unlock_Bootloader>adb shell getprop ro.boot.flash.locked
1

TomKing062, 2023-11-06:
#phone
dd if=uboot_unlock.bin of=/dev/block/by-name/uboot
#pc
spd_dump fdl fdl1-sign.bin 0x5000 fdl fdl2-sign.bin 0x9efffe00 exec erase_part splloader reset
spd_dump exec_addr 0x4ee8 fdl spl_unlock.bin 0x5000
spd_dump fdl spl_bak.img 0x5000
spd_dump fdl spl_bak.img 0x5000
#phone
dd if=/dev/block/by-name/uboot_bak of=/dev/block/by-name/uboot
dd if=spl_bak.img of=/dev/block/mmcblk0boot0
dd if=spl_bak.img of=/dev/block/mmcblk0boot1

Skorpion96, 2023-11-06:
[commands.txt](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/13269886/commands.txt)
mmcblk0boot0 and mmcblk0boot1 are ro as before, also fdl2 gives issues as before. Maybe this phone can't be unlocked by this method? I really don't know but if it's that the case i am really unlucky. I'll leave the eng firmware if you want to have a look at it:https://androidhost.ru/Y56

TomKing062, 2023-11-07:
seems erase_part splloader didn't erase successfully

Skorpion96, 2023-11-07:
I pretty much think that splloader didn't erase because fdl2 isn't loaded correctly, I don't know why but on research download tool it is loading but not here on the exploit

Skorpion96, 2023-11-07:
I did read there:https://github.com/ilyakurdyukov/spreadtrum_flash that fdl2 error could be because we have to trigger a format
[image]
Maybe this explains the reason of the error with fdl2

Skorpion96, 2023-11-08:
[terminal output.txt](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/13295141/terminal.output.txt)
So i tried other variants of the commands, including one where i tried to flash an erased splloader i made by erasing manually splloader image file with an hex editor. No effect on bootloader, after presumable flashing it's still locked

TomKing062, 2023-11-08:
I prefer to use remote software to take a look, you can contact my telegram

Skorpion96, 2023-11-08:
Sure, let me contact you on telegram

Skorpion96, 2023-11-08:
[CVE-2022-38694_Unlock_Bootloader_ZTE_Blade_A5_2019.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/13297791/CVE-2022-38694_Unlock_Bootloader_ZTE_Blade_A5_2019.zip)
For who needs unlock files. Yes we unlocked bootloader LMFAO
Instructions inside in interactive.txt
Before those do:
adb push uboot_unlock.bin /sdcard
adb root
adb shell
dd if=/sdcard/uboot_unlock.bin of=/dev/block/by-name/uboot

lanlanpy, 2023-12-06:
> Terminal Output:[commands.txt](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/13269886/commands.txt)
> 
> mmcblk0boot0 and mmcblk0boot1 are ro as before, also fdl2 gives issues as before. Maybe this phone can't be unlocked by this method? I really don't know but if it's that the case i am really unlucky. I'll leave the eng firmware if you want to have a look at it:https://androidhost.ru/Y56

mmcblk0boot0 are ro ,Write permissions need to be turned on. You can try this command in the ADB shell
enable rw  
#echo 0 > /sys/block/mmcblk0boot1/force_ro
disable rw

#echo 0 > /sys/block/mmcblk0boot1/force_ro

Skorpion96, 2023-12-06:
> > Terminal Output:[commands.txt](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/13269886/commands.txt)
> > mmcblk0boot0 and mmcblk0boot1 are ro as before, also fdl2 gives issues as before. Maybe this phone can't be unlocked by this method? I really don't know but if it's that the case i am really unlucky. I'll leave the eng firmware if you want to have a look at it:https://androidhost.ru/Y56
> 
> mmcblk0boot0 are ro ,Write permissions need to be turned on. You can try this command in the ADB shell enable rw #echo 0 > /sys/block/mmcblk0boot1/force_ro disable rw
> 
> #echo 1 > /sys/block/mmcblk0boot1/force_ro

I thank you for your help but I already unlocked the bootloader, but your help will be probably useful to someone else who knows

lanlanpy, 2023-12-06:
> I thank you for your help but I already unlocked the bootloader, but your help will be probably useful to someone else who knows

That would be great. You're welcome.

konstantinkorovajcev, 2025-02-17:
Nevermind. The ZTE Blade A5 2019 one worked!