TomKing062/CVE-2022-38694_unlock_bootloader #5: Hisense a5 pro CC

wood1978, 2023-07-06:


wood1978, 2023-07-06:
USB ID different because into recovery mode not download mode. 

Follow below step can into download mode:
1. Power off the device.
2. Run Zadig on PC and enable "Options->Log Verbosity->info".
3. Plug in usb cable between PC and phone then wait for the screen to show charging battery.
4. Push power + up + down at same time.
5. After Zadig log shows find a new device and install the driver.
6. After driver install run command in readme.txt.

closed 2023-07-06 by wood1978

wood1978, 2023-07-06:
Below list out unlock success process:
D:\hisense_a7cc>spd_dump exec_addr 0x3f88 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec read_part teecfg 0 1M teecfg.bin read_part trustos 0 6M tos.bin read_part sml 0 1M sml.bin erase_part splloader reset
libusb_control_transfer ok
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
current exec_addr is 0x3f88
SEND FDL1
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
KEEP_CHARGE FDL1
SEND uboot-mod.bin
FDL2: incompatible partition
EXEC FDL2
dump_partition: teecfg+0x0, target: 0x100000, read: 0x100000
dump_partition: trustos+0x0, target: 0x600000, read: 0x600000
dump_partition: sml+0x0, target: 0x100000, read: 0x100000
Answer "yes" to confirm the "erase partition" command: yes

D:\hisense_a7cc>chsize teecfg.bin

D:\hisense_a7cc>chsize tos.bin

D:\hisense_a7cc>chsize sml.bin

D:\hisense_a7cc>spd_dump exec_addr 0x3f28 fdl fdl1-boot.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 fdl teecfg.bin 0x9401fe00 fdl tos.bin 0x9403fe00 fdl sml.bin 0x93fffe00 exec
libusb_control_transfer ok
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
current exec_addr is 0x3f28
SEND FDL1
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
KEEP_CHARGE FDL1
SEND uboot-mod.bin
SEND teecfg.bin
SEND tos.bin
SEND sml.bin
usb_recv failed : LIBUSB_ERROR_IO

D:\hisense_a7cc>spd_dump exec_addr 0x3f28 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec erase_part uboot_log write_part splloader spl.bin timeout 100000 write_part userdata userdata.bin reset
libusb_control_transfer ok
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
current exec_addr is 0x3f28
SEND FDL1
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
KEEP_CHARGE FDL1
SEND uboot-mod.bin
FDL2: incompatible partition
EXEC FDL2
Answer "yes" to confirm the "erase partition" command: yes
file size : 0xe854
Answer "yes" to confirm the "write partition" command: yes
load_partition: splloader, target: 0xe854, written: 0xe854
file size : 0x2413c
Answer "yes" to confirm the "write partition" command: yes
load_partition: userdata, target: 0x2413c, written: 0x2413c

Xinon232, 2023-07-06:
Would this be applicable to the Hisense A5 Pro CC as well?

wood1978, 2023-07-07:
> Would this be applicable to the Hisense A5 Pro CC as well?

Sorry, Only can confirm A7CC work.

xucongyong, 2023-07-31:
Push power + up + down at same time.
not into download mode

wood1978, 2023-08-01:
> 
Need to check the driver install first then you will see the device in download mode by Zadig log.