TomKing062/CVE-2022-38694_unlock_bootloader #8: Impossible to create boot.img for Hisense A5PRO CC with avbtool

Xinon232, 2023-07-11:
`python2 C:\Python27\Tools\Scripts\avbtool.py add_hash_footer --image image-new.img --partition_name boot --partition_size 36700160 --key rsa4096_vbmeta.pem --algorithm SHA256_RSA4096 --salt 5F55215FD2302D021F850B55912ED48D176784678692DC012E054B1ECD0BE025
Traceback (most recent call last):
  File "C:\Python27\Tools\Scripts\avbtool.py", line 4417, in <module>
    tool.run(sys.argv)
  File "C:\Python27\Tools\Scripts\avbtool.py", line 4253, in run
    args.func(args)
  File "C:\Python27\Tools\Scripts\avbtool.py", line 4310, in add_hash_footer
    args.do_not_use_ab)
  File "C:\Python27\Tools\Scripts\avbtool.py", line 3041, in add_hash_footer
    image = ImageHandler(image_filename)
  File "C:\Python27\Tools\Scripts\avbtool.py", line 697, in __init__
    self._read_header()
  File "C:\Python27\Tools\Scripts\avbtool.py", line 719, in _read_header
    _) = struct.unpack(self.HEADER_FORMAT, header_bin)
struct.error: unpack requires a string argument of length 28`

I already followed step 0,1,2 of the Tutorial [ Android 10: https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/wiki/Magisk ]

but due to the error I can't create a final boot.img for root.

I uploaded the original boot.img and the repacked&magisk patched image-new.img here:
https://filebin.net/alcq7svzcl3oso9b

trainingdummy, 2023-07-11:
Pretty sure you just need to make sure all your tools (python2, avbtool, etc.) are in the same place. They won't run if they're in a different folder than the directory you're running the command prompt from.

Xinon232, 2023-07-11:
no it doesn't work...

`C:\WINDOWS\system32>python2 C:\Python27\avbtool add_hash_footer --image image-new.img --partition_name boot --partition_size 36700160 --key rsa4096_vbmeta.pem --algorithm SHA256_RSA4096 --salt 5F55215FD2302D021F850B55912ED48D176784678692DC012E054B1ECD0BE025
usage: avbtool add_hash_footer [-h] [--image IMAGE]
                               [--partition_size PARTITION_SIZE]
                               [--partition_name PARTITION_NAME]
                               [--hash_algorithm HASH_ALGORITHM] [--salt SALT]
                               [--calc_max_image_size]
                               [--output_vbmeta_image OUTPUT_VBMETA_IMAGE]
                               [--do_not_append_vbmeta_image]
                               [--algorithm ALGORITHM] [--key KEY]
                               [--signing_helper APP]
                               [--signing_helper_with_files APP]
                               [--public_key_metadata KEY_METADATA]
                               [--rollback_index ROLLBACK_INDEX]
                               [--append_to_release_string STR]
                               [--prop KEY:VALUE] [--prop_from_file KEY:PATH]
                               [--kernel_cmdline CMDLINE]
                               [--setup_rootfs_from_kernel IMAGE]
                               [--include_descriptors_from_image IMAGE]
                               [--print_required_libavb_version]
                               [--chain_partition PART_NAME:ROLLBACK_SLOT:KEY_PATH]
                               [--flags FLAGS] [--set_hashtree_disabled_flag]
                               [--use_persistent_digest] [--do_not_use_ab]
avbtool add_hash_footer: error: argument --image: invalid FileType('rab+') value: 'image-new.img'`

neither does it work if i put it directly in the folder. the terminal doesn't understand that the file is right there

Xinon232, 2023-07-12:
If anyone is able to create a final boot img for root, uploading it would be highly appreciated.

TomKing062, 2023-07-13:
https://files.catbox.moe/wak081.img

[image]

closed 2023-07-13 by TomKing062

FFFKE, 2023-10-22:
THIS FOR A5 PRO CC BOOT,IME ?

tiubenedict, 2024-07-19:
> THIS FOR A5 PRO CC BOOT,IME ?

sorry necro, but yes for a5 pro cc