TomKing062/CVE-2022-38694_unlock_bootloader #9: Fastboot commands won't run properly

trainingdummy, 2023-07-11:
I'm currently stuck trying to get `fastboot` commands to work properly. I've successfully booted into fastbootd (yes, with a d at the end) with the `adb reboot fastboot` command, but at that point, none of the following commands work. The only response I get with them is `< waiting for any device >`. Not sure why it can't find it yet, but I've been unable to run anything with `fastboot` throughout this whole process. Any ideas why that might be?

trainingdummy, 2023-07-11:
Just noticed that I needed to run Zadig again and reinstall the libusbK driver, since it seemingly got cleared during the factory reset. Didn't change anything, but may have been related to the issue.

wood1978, 2023-07-11:
fastboot and adb need to use google USB driver.
So you need to check the driver by [DriverStoreExplorer](https://github.com/lostindark/DriverStoreExplorer/releases).

google USB driver:
https://developer.android.com/studio/run/win-usb

trainingdummy, 2023-07-11:
Okay, I think I installed it on the Unisoc Phone entry, but now it can't even detect the device with `adb devices`. What did I miss?

trainingdummy, 2023-07-11:
When viewing it in Zadig again, while on charging mode only, it shows Hisense Phone and a WINUSB driver. When it's put into MTP mode, though, it splits into two drivers, one under ADB Interface (WINUSB) and one under MTP (WUDFWpdMtp).

wood1978, 2023-07-12:
The A7CC Zadig info list is below:
ADB interface: Driver USBSTOR(v10.0.22000.918)
one under MTP: WUDFWpdMtp(v10.0.22000.918)

TomKing062, 2023-07-13:
search and install SPD Driver R4.20

closed 2023-07-13 by TomKing062

trainingdummy, 2023-07-15:
It didn't do anything, as far as I can tell. Same result, same two inputs.