TomKing062/CVE-2022-38694_unlock_bootloader #11: Phone is rooted but Google Services are still not usable (Hisense A5 Pro CC)

Xinon232, 2023-07-13:
With the boot.img from TomKing062 I rooted the phone successfully.

However getting Google Services to work is another story unfortunately.
Even with root, Google Services are still not accessible.

I followed a tutorial to integrate google apps into the system: https://blindhelp.net/blog/how-install-google-apps-without-custom-recovery-learn-manually-install-gapps-any-chinesenon

However even with root I can't modify the hosts file (system/etc) to remove the Google-Server block. 
Integrating the apps into the system also doesn't work as adb tells me that there is "no space left on device" which is not really true...

TomKing062, 2023-07-13:
use gapps magisk module

closed 2023-07-13 by TomKing062

Xinon232, 2023-07-13:
litegapps gave me a bootloop