TomKing062/CVE-2022-38694_unlock_bootloader #12: Phone is rooted but Google Services softbrick my phone (Hisense A5 Pro CC bootloop)

Xinon232, 2023-07-14:
I tried to install Google Services using the Magisk-module LiteGapps but it gave me a bootloop. 

Is there another Module which works with the phone? There doesn't seem to be too much around for Android 10...

TomKing062, 2023-07-15:
[image]

disable avb by flash vbmeta then modify the blacklist or create a empty blacklist manually in any magisk module

wood1978, 2023-07-15:
My A7CC use WeebGapps work fine for you reference: https://t.me/WeebGAppsChannel/662

lichobg, 2023-07-15:
@ Xinon232
Hi, see https://forum.xda-developers.com/t/hisense-a5-pro-cc-how-to-root-unlock-bootloader.4109021/page-10 - 

Install Magisck&Google play Part 2. Download signet&patched image for your Phone, install it in fastboot mode and install Magisck&Google play as modul.

closed 2023-07-23 by TomKing062