TomKing062/CVE-2022-38694_unlock_bootloader #14: Can you add the tutorial (on the wiki) more detail, or patch my bootloader (will give the GDrive link)??

Prowindows62, 2023-07-27:


Scarletya, 2023-07-28:
Yes please 

(normal download) send fdl1-dl.bin, send uboot, write uboot-mod to uboot, erase splloader

(fallback download) send spl-custom.bin

(fallback download) send fdl1-dl.bin, send uboot, write official uboot back, write official splloader back, erase userdata.

Someone did make a easy guide for FDL method, but spl method still does have any guide....

I really tried to understand and tried to find how to type commend but no....

'write uboot-mod to uboot, erase splloader'

=> If someone dont mind please give us more detail commend 

A5 pro cc A7 A7cc are now someone made a quick guide but its only about fdl method so I dont get how to do with spl method 

Please help

TomKing062, 2023-07-29:
newer uboot has extra check when write uboot-mod to uboot partition, and when use spl method, the `reboot_devices` func randomly bring device to `normal download`

and fdl1 method sometimes face a timeout problem which spl won't have, maybe i should change chsize code (low priority)

closed 2023-08-10 by TomKing062