TomKing062/CVE-2022-38694_unlock_bootloader #16: 有關 Magisk BOOT.IMG

sulunlun, 2023-08-31:
sign patched boot with avbtool or you will stuck at bootlogo
不能直接安装，必须选择文件修补，然后用avbtool签名，签名后的boot才不会卡开机

請問如果我把自己存檔的 BOO.img 用在另一支Magisk PATH的 後的存檔, 可以直接刷入嗎?!
還是要照您說的需要用avbtool签名 , 才能刷入?

If I use the BOO.img of my own archive in another Magisk PATH archive, can it be flashed directly?
Or do I need to sign it with avbtool as you said?

sulunlun, 2023-08-31:
謝謝回覆，所以用中文可以交流嘛
我在準備liunx環境了，準備來作簽章

sulunlun, 2023-09-01:
我手機是 海信A5PRO
目前已經解鎖了 , 但是卡在要把boot.img下載加簽 ., 請問出現下列
usb_send failed : LIBUSB_ERROR_TIMEOUT
要怎麼處理 
PS:有換掉 USB線了

I have unlocked the bootloader.
and now I'm downloading boot.img
I keep getting the following message , how do I troubleshoot it?

D:\hisense\A5Root>spd_dump exec_addr 0x3ee8 fdl fdl1-dl.bin 0x3ee8 fdl uboot-mod.bin 0x9efffe00 exec read_part boot 0 35M boot.img reset
libusb_control_transfer ok
CHECK_BAUD FAIL
CHECK_BAUD FAIL
CHECK_BAUD FAIL
usb_send failed : LIBUSB_ERROR_TIMEOUT

sulunlun, 2023-09-01:
用ADB 進入 , 就可以下載了
adb reboot autodloader

eveing, 2023-09-04:
> 我手機是 海信A5PRO 目前已經解鎖了 , 但是卡在要把boot.img下載加簽 ., 請問出現下列 usb_send failed : LIBUSB_ERROR_TIMEOUT 要怎麼處理 PS:有換掉 USB線了
>
> I have unlocked the bootloader. and now I'm downloading boot.img I keep getting the [...]

我是A5Pro标准版(NFC)，exec_addr 0x3f48成功读取boot.img，你试试

sulunlun, 2023-09-05:
> 我是A5Pro标准版(NFC)，exec_addr 0x3f48成功读取boot.img，你试试

謝謝您 我後來沒有下載boot.img , 直接刷人PATH好的boot.img , 目前已ROOT 好也刷入GOOGLE 服務, 
[eveing](https://github.com/eveing) 可以請問一下 您聯絡人同步正常嗎?!

我作過以下動作
我刷[MAGISK]LiteGapps_arm64_10.0_v2.9_official , 目前只剩下聯絡人沒辦法同步
如果刷joy_gapps_installer-R07-core , 就不開機
用Google_Installer_3.0 沒反應

Thank you I didn't download the boot.img, I just flashed the boot.img from the PATH, and now it's ROOTed and flashed to GOOGLE service, so I'm not sure if it's working. 
[eveing](https://github.com/eveing) Can you tell me if your contacts are synchronized properly?
I did the following
I flashed [MAGISK]LiteGapps_arm64_10.0_v2.9_official , and now only my contacts are not synchronized.
If I use joy_gapps_installer-R07-core , it won't boot.
Google_Installer_3.0 doesn't work.

eveing, 2023-09-05:
我这里用不到谷歌服务的，这个问题没法回答你了。

sulunlun, 2023-09-06:
> 我这里用不到谷歌服务的，这个问题没法回答你了。

謝謝您 我再試試

closed 2023-09-09 by TomKing062