TomKing062/CVE-2022-38694_unlock_bootloader #19: Imposible to upgrade from Magisk 26.1 to Magisk 26.3

ferlanero, 2023-09-05:
Making an automatic upgrade from Magisk 26.1 to Magisk 26.3 leads to a phone brick.

Is there any way to upgrade Magisk without loosing phone data?

TomKing062, 2023-09-06:
sign process is same

after automatic upgrade, don't reboot
`dd if=/dev/block/by-name/boot of=/sdcard/boot.img`
sign
`dd if=/sdcard/boot_signed.img of=/dev/block/by-name/boot`
now reboot will be safe

ferlanero, 2023-09-06:
> sign process is same
> 
> after automatic upgrade, don't reboot `dd if=/dev/block/by-name/boot of=/sdcard/boot.img` sign `dd if=/sdcard/boot_signed.img of=/dev/block/by-name/boot` now reboot will be safe

Commands must be entered in a console like Termux?

TomKing062, 2023-09-06:
yes, termux, adb shell …

ferlanero, 2023-09-08:
> sign process is same
> 
> after automatic upgrade, don't reboot `dd if=/dev/block/by-name/boot of=/sdcard/boot.img` sign `dd if=/sdcard/boot_signed.img of=/dev/block/by-name/boot` now reboot will be safe

Ok, I understand. But which is the process to sign boot.img?

TomKing062, 2023-09-09:
magisk patched boot is still 35M, so avbtool will give size error
goal of unpack and repack is to decrease image size

ferlanero, 2023-09-09:
> magisk patched boot is still 35M, so avbtool will give size error goal of unpack and repack is to decrease image size

Thanks. I´m trying to unpack & repack with AIK boot.img which I get from `dd if=/dev/block/by-name/boot of=/sdcard/boot.img` but when I do `unpackimg.bat boot.img` AIK gives me this error:

```
Android Image Kitchen - UnpackImg Script
by osm0sis @ xda-developers

Supplied image: boot.img

Setting up work folders . . .

Unrecognized format.
Error!
```

I can´t continue with the process. Do you have any advice, please?

closed 2023-09-25 by TomKing062