TomKing062/CVE-2022-38694_unlock_bootloader #20: How can I get the stack info from another chipsets

zzxx3081, 2023-09-12:
Dear TomKing,

Thank you! Your advice was very helpful.
However, There are some parts I don't understand. I would like to know how your csv* came out.

*CVE-2022-38694_unlock_bootloader/ums512/stack-info-ums512.csv

I'm interested in your data, so I'm trying it out on my own chipset.
It is "Unisoc SC9863A".

I like your materials so much that my studies are going well.
My goal is to complete your tutorial(UNLOCK FDL1 method) targeting this chipset.
I desperately need your help.

Best regards!

TomKing062, 2023-09-13:
get stack info by analyzing BootROM in IDA

for get BootROM, **device need to be not fused**, and we need do some modification to fdl1, (by compile or directly change asm code)

to compile, read [dump_mem](https://github.com/TomKing062/jing_patch/tree/dump_mem)
for direct  assembly language, i will a example later

TomKing062, 2023-09-13:
[dump_ums9620.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/12599902/dump_ums9620.zip)

zzxx3081, 2023-09-14:
I try it. Thank you!

closed 2023-09-25 by TomKing062

TomKing062, 2023-11-05:
9863a added now