TomKing062/CVE-2022-38694_unlock_bootloader #21: unlocking guide for real life required

piter2100, 2023-09-21:
Hello Sir, 
is it possible to add any appropriate "how to" tutorial for avarage pc user with basic hardware knowledge ?
So I could finaly perform a bootloader unlocking on my nokia t20 with unisoc ums512.
Thanks

TomKing062, 2023-09-22:
i did want to, until i found memory initialization cannot be universal, (or it can but i don't have any datasheet to know the meaning of initialization parameters, so i just copy everything from splloader to source code )

for nokia t20, i remember there is a fdl1 in update package can be used, but i still need to find the modification point in uboot for a11(+), **MAYBE** next month.

TomKing062, 2023-09-24:
found a new modify point and tested good on my ud710, pm me by https://t.me/QuTick102

closed 2023-10-05 by TomKing062