TomKing062/CVE-2022-38694_unlock_bootloader #26: 海信F50+(HDR551T)是否有解锁可能？

chinese-wzq, 2023-10-28:
[image]
如图，如果有解锁可能的话我可以自己尝试解锁

TomKing062, 2023-10-29:
7510全都能解，我看信封没给551的完整包，把2020的bat替换成下面的试试（已经加上备份恢复spl了，建议中间按任意键继续的时候复制到别处去），没解的话再修改uboot，在miscdata的0x2000-0x2040是解锁信息
```
spd_dump exec_addr 0x3f88 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec read_part trustos 0 6M tos.bin read_part sml 0 1M sml.bin read_part splloader 0 1M spl_bak.bin erase_part splloader reset
chsize tos.bin
chsize sml.bin
chsize spl_bak.bin
TIMEOUT /T 10 /NOBREAK
spd_dump exec_addr 0x3f28 fdl fdl1-boot.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 fdl tos.bin 0x9403fe00 fdl sml.bin 0x93fffe00 exec
TIMEOUT /T 10 /NOBREAK
spd_dump exec_addr 0x3f28 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec verbose 2 read_part miscdata 8192 64 m.bin reset
pause
spd_dump exec_addr 0x3f28 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec erase_part uboot_log write_part splloader spl_bak.bin reset
```

chinese-wzq, 2023-11-07:
好，我这几天就解锁试试！争取成为支持列表中的设备好吧（希望不是失败案例）

Liusx1024, 2023-11-13:
兄弟解锁成功了吗

chinese-wzq, 2023-11-18:
> 兄弟解锁成功了吗

这几天很忙，尽量找时间吧

TomKing062, 2024-01-04:
传了

closed 2024-01-04 by TomKing062