TomKing062/CVE-2022-38694_unlock_bootloader #29: Compatibility with ZTE Blade V40 Design

justasierof, 2023-11-11:
I want to ask if it's possible to make compatible the ZTE Blade V40 Design with bootloader unlock. I can send firmware if it's needed

justasierof, 2023-12-11:
The chip that has is UNISOC T616

TomKing062, 2023-12-15:
fdl1 fdl2 u-boot-spl-16l-ufs

Skorpion96, 2023-12-17:
he is a bit screwed as that phone has no firmware online, but as i'm doing on my ZTE Blade A52 he can try to find a dump and use fdl1 from another model with the same processor

Skorpion96, 2023-12-18:
[uboot.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/13709336/uboot.zip)
here uboot (fdl2) for ZTE Blade V40 Design
for fdl1 we can get it from another device with the same processor from his firmware

TomKing062, 2024-07-22:
try [test.ums9230_universal_unlock.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/releases/download/1.71/test.ums9230_universal_unlock.zip)

TomKing062, 2024-10-02:
since ums9230_universal_unlock works for ZTE Blade V50 Design (ums9230)
i am closing this

closed 2024-10-02 by TomKing062