TomKing062/CVE-2022-38694_unlock_bootloader #31: Compatibility questions with KDXF T10（ud710_2h10,Android 9）

qwqlemon2333, 2023-11-24:
Hello, I encountered some issues while modifying uboot. It seems that the manufacturer has rebooted it_ Functions like devices are all hidden, which makes me unsure how to modify them to forcibly unlock the bootloader. If possible, could you take a look? I have uploaded some files that I extracted from the upgrade package

您好，我在修改uboot时出现了一些问题，厂商貌似把reboot_devices之类的函数都给隐藏了，导致我也不知道该怎么修改来强制解锁bootloader，如果行的话您能研究一下吗，我上传了从升级包中提取出的spl和uboot文件

[uboot-and-spl.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/13461514/uboot-and-spl.zip)

TomKing062, 2023-11-25:
do_download 9F005E2C
dl_reboot 9F017ABC
reboot 9F02AE08
reboot_0 9F02AF78
dl_poweroff 9F017AE0
poweroff 9F02AEBC
setlock 9F066274
getlock 9F066370
do_cboot 9F0060FC

一个都没少，而且吧，你这是个紫光公钥啊，直接签identifier.bin就解锁了（）哪用得上这个漏洞

TomKing062, 2023-11-25:
光记着do_download的0x17是dl_poweroff了，do_download的0x5就是dl_reboot我怎么忘了（），我之后截点图放进wiki去2333

closed 2023-11-25 by TomKing062

TomKing062, 2023-11-25:
再补一个
_check_secure_part 9F0653C4

qwqlemon2333, 2023-11-26:
but,有个更奇怪的问题，为啥解锁了但boot,vbmeta刷都刷不进去，boot分区直接一尝试刷就直接重启了

TomKing062, 2023-11-26:
boot用[rsa4096_boot.pem](https://gitlab.com/unisoc-android/android_vendor_sprd/-/blob/W20.13.3/proprietories-source/packimage_scripts/signimage/sprd/config/rsa4096_boot.pem?ref_type=heads)签过了么
或者用delta的system模式root
vbmeta要刷也要签名

qwqlemon2333, 2023-12-02:
仍然有问题，我这个设备是lagacy SAR的分区架构，得patch rec，然后我patch完再用你上面提供的.pem密钥签名并刷入（fastboot下），但仍然刷不进去

TomKing062, 2023-12-02:
boot是boot，rec是rec，要看vbmeta的，要不就把整个vbmeta重造了，多看看hovatek（）
我遇到过补rec无效的，不然我怎么老说delta改system

qwqlemon2333, 2024-02-17:
大佬，delta改system

qwqlemon2333, 2024-02-17:
大佬，delta改system有脚本了吗

sdgasdgahj, 2024-02-23:
ok，我成功root了