TomKing062/CVE-2022-38694_unlock_bootloader #34: REALME GT2 RMX3311 QUALCOMM CHIP AND REALME C11 2023 RMX3231 UNISONIC CHIP

NKMyoutube, 2023-12-05:
CAN YOU CREATE BOOTLOADER UNLOCKER FOR REALME GT2 RMX3311 QUALCOMM CHIP AND REALME C11 2023 RMX3231 UNISONIC CHIP?

team-orangeBlue, 2023-12-05:
你以下業務出現違規異常，了解詳情請按0， 由華語客服為你服務

NKMyoutube, 2023-12-05:
你能創建引導程式解鎖 Realme C11 2021 RMX3231 UNISONIC 晶片嗎

qwqlemon2333, 2023-12-07:
In fact,you can follow the link below:[How to unlock Unisoc (SPD) bootloader using Identifier Token](https://www.hovatek.com/forum/thread-32287.html)

qwqlemon2333, 2023-12-07:
> In fact,you can follow the link below:[How to unlock Unisoc (SPD) bootloader using Identifier Token](https://www.hovatek.com/forum/thread-32287.html)

It's the general way to unlock the Unisoc bootloader

qwqlemon2333, 2023-12-07:
If you want to unlock by this Project,you need your device's stock firmware(a .pac file) and extract the fdl1,fdl2(uboot),spl(splloader),tos(trustos),sml,teecfg images via [Extract Images from .pac file from Spreadtrum Unisoc SPRD](https://github.com/NasdaqGodzilla/UNISOC_SPRD_PAC_UNPAC) then use [The Wiki's guide](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/wiki/AddSupportToModel)

closed 2023-12-12 by TomKing062