TomKing062/CVE-2022-38694_unlock_bootloader #37: Alldocube iPlay50 Mini(T811)

desugar-64, 2023-12-24:
I used your ums9230_alldocube50mini unlocking tool to unlock the bootloader on my Alldocube iPlay 50 mini (T811) tablet. It kind of worked - during boot up there is now a warning message about the device being unlocked, and the tablet itself still functions properly. However, when I put the tablet into bootloader mode, I am unable to flash a new vbmeta image. The writing process just hangs and does nothing.

I also tried installing the AOSP 13 and 14 using theDSU and Shizuku/ADB tools. The installation process reported completing successfully, but after rebooting, the tablet fails to load the B system partition and instead falls back to the A partition. During boot up, I see a black screen before the tablet reboots back into the original stock firmware.

This makes me think that maybe the bootloader is not fully unlocked, if that is possible.

desugar-64, 2023-12-26:
nvm...

closed 2023-12-26 by desugar-64

Fyramiz, 2024-05-17:
- first, don't use dsu with unisoc
Second flash images in fastbootd to enter use: adb reboot fastboot
Also don't flash vbmeta in unisoc, it will cause bootloop, however flashing GSI's(lineage TD, evolutionX TD(TD = Treble-Droid)) without vbmeta will work normally

Fyramiz, 2024-05-17:
And don't use DSU with unisoc