TomKing062/CVE-2022-38694_unlock_bootloader #38: Hisense A7 soft bricked, please help, thx!

daopeng, 2023-12-29:
I have a Hisense A7 running on Android 10, and I first unlocked bootloader on Windows 11 with no problems, but then stuck at extracting boot.img. The terminal returns 'bad command args' error and so I searched closed issues list to find a solution untill stumbled upon using 'adb reboot autodload' then desperately runned it only to find my phone freezed and unresponsive. I then checked #6 and got to the step of installing libusbK using Zadig but not sure what to do next, the 3 commands seemed to be phone specific. Please advise what to do to save the phone. Thanks a lot.

TomKing062, 2024-01-01:
there is a splloader(unmodified) in unlock tool folder, the command to write back is
```
spd_dump exec_addr 0x3f28 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec write_part splloader spl.bin reset
```

daopeng, 2024-01-01:
Thanks so much for your reply. Could you also elaborate which driver to use in Zadig and how to enter download mode to execute the above command? The thing is, I have tried both WinUSB and libusbK, pressing power+up+down combo, and the phone is still unresponsive, the terminal throws `Waiting for connection` and `find port failed` error. Thanks again!

TomKing062, 2024-01-01:
uploaded libusb spd_dump to release

daopeng, 2024-01-01:
Thanks so much. I downloaded it and find libusb and SPRD folder, after comparing the date of `spd_dump.exe`, I used `libusb` folder as the working folder and also copied `Channel9.dll` from the other one into it, but the terminal gave me `custom_exec_no_verify_3f28.bin does not exist.` error. However, I used spd_dump_interactive.exe and could get to the BROM> prompt, so things are definitely going to the right direction. What shall I do next, thanks!

TomKing062, 2024-01-01:
extract to ud710_hisense_a7.zip folder (replace packed SPRD ver)

daopeng, 2024-01-01:
It worked! Thanks sooooo much! You're the best!

daopeng, 2024-01-01:
My phone is saved and it's been great! Thanks so much! Now I have the bootloader unlocked, Magisk installed, but still couldn't extract the boot.img with the updated `spd_dump.exe` and `libusb` driver, using the original address `0x3f88` (not the fallback address `0x3f28` as directed) with the command `spd_dump exec_addr 0x3f88 fdl fdl1-dl.bin 0x3f88 fdl uboot-mod.bin 0x9efffe00 exec read_part boot 0 35M boot.img reset`, the terminal gave me `timeout reached` error. Could you help please? Thanks a lot.

TomKing062, 2024-01-01:
`spd_dump exec_addr 0x3f88 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec read_part boot 0 35M boot.img reset`

daopeng, 2024-01-01:
I'll give it a try and many, many thanks!

daopeng, 2024-01-01:
Voila, it runs smoothly! Now I have the boot.img! Thanks sooooo much, really appreciate it. I'll go on with the rest steps then.

daopeng, 2024-01-01:
I'm happy to report back: all done. rooted! Thanks a lot man, really appreciate your help!

closed 2024-01-01 by daopeng