TomKing062/CVE-2022-38694_unlock_bootloader #41: Support for Umidigi G1 Max

jonathanrobichaud4, 2024-01-09:
Hi, I recently purchased a Umidigi G1 max but learned that unlocking the bootloader is a pain. It uses a T610. I used 
the Alldocube iplay 50 files and it kind of worked so I think it's possible. (It didn't work but I was still able to reflash the original firmware so I didn't break it) I have the .pac file for the firmware and the logs from running the files mentioned above. I can provide them if needed but I am also not too sure where to start!

TomKing062, 2024-01-09:
fdl1 fdl2 u-boot-spl-16k

jonathanrobichaud4, 2024-01-09:
Here they are:
[Umidigi Files.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/13878629/Umidigi.Files.zip)
Please let me know if I need to do anything. I tried opening them in ghidra to decompile them but I'm not super sure how to use it.

jonathanrobichaud4, 2024-01-10:
Ok I have fd1 opened in ida and I can follow along with the first tutorial and recognize what to look for but it is kind of different. I'm not sure how to explain it. I'm not sure if some of my addresses or something are incorrect. But, I think it bodes well that I can open it?

jonathanrobichaud4, 2024-01-10:
[image]
Screenshot for reference of the (i think?) verification.

TomKing062, 2024-01-11:
done

closed 2024-01-11 by TomKing062