TomKing062/CVE-2022-38694_unlock_bootloader #47: Hisense Q5 stuck, "chsize spl.bin" returned "The file is not sprd trusted firmware"

Ranomez, 2024-01-24:
Hello!
I attempted unlocking the bootloader of my Q5 using this tool but now can't exit out of the download mode.
While the batch file was executing I did see that "chsize spl.bin" returned "The file is not sprd trusted firmware" so I'm guessing it was a bad dump but since the first command already erased sploader I couldn't really do much anymore.
What can I do in this situation to fix my tablet? Would I need a spl dump from someone else? If so how could I get one?

Ranomez, 2024-01-24:
Extra details: checking the spl.bin file in a hex editor shows it's empty.

Ranomez, 2024-01-24:
Oh my god! Thank you so much!
That info should really be added to the wiki though.

closed 2024-01-24 by Ranomez

Ridhoardhiansyah7, 2025-04-24:
I experienced the same issue when trying to unlock the bootloader on the Narzo 50i. If I’m not mistaken, after the process of erasing splloader and splloader_bak, there was an instruction to press any key to continue. I stopped there and unplugged the USB cable, then repeated the unlock bootloader process. It ended with the error 'The file is not sprd trusted firmware.' Can I use this method?

.\spd_dump exec_addr 0x4ee8 fdl fdl1-dl.bin 0x5000 fdl fdl2-dl.bin 0x9efffe00 exec erase_part uboot_log write_part splloader u-boot-spl-16k-sign.bin timeout 100000 erase_path userdata reset

TomKing062, 2025-04-24:
@Ridhoardhiansyah7 grub splloader from Narzo 50i pac firmware

SEM0001, 2025-04-27:
@TomKing062 Hey, please correct me if I'm wrong, but are the spl dumps no longer included with the unlock releases? Would you be willing to share the older versions of the files again?

There are some guides online that tell you to run adb reboot autodloader, and from what I understand, that always causes a soft brick and you have to flash the splloader back. This makes things pretty tricky for people following those guides.

TomKing062, 2025-04-27:
i don't keep former releases
(dump from devices + gen_spl-unlock(-legacy)) works better then (pre-include spl-sign + spl-unlock)