TomKing062/CVE-2022-38694_unlock_bootloader #50: "fopen (load) failed"

Educg96, 2024-01-31:
Hello @TomKing062 , thank you for the work that you've been doing. After inserting the next command:

`spd_dump exec_addr 0x3f28 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec erase_part uboot_log write_part splloader spl.bin timeout 100000 write_part userdata userdata.bin reset`

I get that message error "fopen (load) failed" 

In the previous command `spd_dump exec_addr 0x3f28 fdl fdl1-boot.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 fdl teecfg.bin 0x9401fe00 fdl tos.bin 0x9403fe00 fdl sml.bin 0x93fffe00 exec`

I got the message "usb_recv failed: LIBUSB_ERROR_IO" but I continued with the next command because I saw that other user got it to work.

Everything else seemed to work fine and I got the boot.img file in the carpet already the phone just doesn't restart normally so I can install Magisk and patch the image. Do you know anything about those errors? I tried copying every different file in the same carpet and using different drivers.

Thank you in advance!

TomKing062, 2024-02-14:
seems like command for ud710 device
forget old guides and always use command in latest unlock.bat

TomKing062, 2024-02-23:
lost splloader backup (including autodloader) has a workaround in batch/shell now

closed 2024-02-23 by TomKing062