TomKing062/CVE-2022-38694_unlock_bootloader #53: Bricked Hisense A5 Pro after Failed Root

kenwang92, 2024-02-04:
Hello everyone,

I attempted to root my Hisense A5 Pro, but unfortunately, the process went awry, and now my device is bricked. I used ```ums512_hisense_a5pro``` for rooting, and my operating system version is the latest.

During the root process, I encountered the following issues:

I have tried the following solutions without success:

I followed the instructions in an [article](https://codeniz.com/2023/10/07/hisense-a5-pro-root-%E8%88%87-google-%E6%9C%8D%E5%8B%99/), and when it came to entering commands, I attempted the ones provided in the article. However, the command prompt displayed ```custom_exec_no_verify_3f48.bin does not exist.``` Subsequently, I ran ```unlock.bat``` and, after holding down the volume up and power buttons for 10 seconds, connected the phone. After inputting several "yes" responses, the command prompt window closed following the execution of the last command. Unfortunately, my phone showed no response. I have tried repeating the steps mentioned above, including holding down the volume up and power buttons, but the phone remains unresponsive. I also attempted to access the 9008 mode by removing the back cover, but I couldn't locate any touchpoints near the flashlight.

Could anyone provide guidance or suggestions on how to recover my Hisense A5 Pro?
Thank you for your assistance!
sorry for my poor english,this is translate by chatgpt
[image]
[image]
[image]
[image]
After executing the last line mentioned above, the terminal closed on its own.

TomKing062, 2024-02-04:
see #47

kenwang92, 2024-02-05:
Thank you very much. My phone has finally entered the system.

closed 2024-02-05 by kenwang92