TomKing062/CVE-2022-38694_unlock_bootloader #57: 小度青禾学习手机V20

weimzh, 2024-02-13:
https://wap.zol.com.cn/1611/1610459/index.html

该手机使用T616芯片, 系统屏蔽了adb fastboot等功能

closed 2024-02-14 by TomKing062

qwqlemon2333, 2024-02-18:
会进下载模式吗

weimzh, 2024-03-01:
xps06e:/ $ getprop ro.build.version.release
12

安卓版本为12，可以在wiki补充一下

weimzh, 2024-03-01:
在QQ群里发过的在这里记录下，以免丢失：

允许安装APK:
root后使用命令 setprop persist.duershow.sys.enable.only_self_update false

允许USB ADB:
用spd_dump读取miscdata分区，搜索600WW，改成其它的如600CN，刷回

Arvintian, 2024-06-26:
怎么进下载模式？

YelZhang, 2026-10-17:
感谢 @TomKing062 和 @weimzh 的工具及记录。补充一台青禾 V20 的成功实测，尤其回答上面的“怎么进下载模式”。核心漏洞、适配、SKU 和安装限制属性均来自社区，这里仅补充本机连接细节与收尾经验。

### 测试范围

XD-SEE00-2301 / xps06e，UNISOC T616，工具识别 UFS、A/B，Android 12，固件 `DuerShow_T616_v1.65.0.20250825010819358.R`。Intel Mac，ADB 35.0.1，设备包来自 release 1.72，spd_dump 从项目源码本地编译。仅单台设备验证，不保证其他批次适用。

### 下载模式：本机可重复步骤

1. 电脑先启动 spd_dump 等待设备。
2. 手机完全关机、拔线，按住音量减，再插入连接电脑的 USB 线，本机保持约 15 秒。
3. 屏幕可能出现品牌文字，随后回到充电图标，不能仅凭图标判断失败。
4. Mac 捕获 VID:PID `1782:4D00`，名字可能为空或 `Gadget Serial`。工具收到 `SPRD3` 并完成 `CMD_CONNECT bootrom` 才确认握手。端口可能短暂出现，工具应预先等待。

连接超时时，退出旧工具、恢复正常启动后再关机重试有效。不要并行运行多个占用 USB 的工具。读取后尽量保留同一个交互会话，避免重新握手失败。

### USB 与安装限制是不同阶段

- 本机最初工程菜单“USB接口激活”要求插入 SIM。插卡并操作后，才有文件传输和 USB 枚举；是否必须联网或付费未验证。
- 初始工程入口 `*#*#83781#*#*`，DEBUG&LOG 中 USB Debug 位于 Watch Dog 与 PreCheck 之间。打开后仍只有 unauthorized，没有授权弹窗。
- 按本 issue 的既有方法，备份 miscdata 后把唯一的 `600WW` 改为同长度 `600CN`，回读核对，随后开机出现 USB 调试授权。不要套用其他机器的固定字节偏移。
- ADB 能安装应用不代表手机本地安装开放。应用宝已允许安装未知应用，仍报“不支持三方渠道安装”；换浏览器、文件管理器或处理微信的 `.apk.1` 后缀不能解决该系统策略。

### 解锁、恢复出厂与调试菜单消失

使用青禾专用包并按上游分阶段流程操作，SPL 解锁补丁由本机原件生成。先备份，结束前恢复本机主备 SPL 与原机 U-Boot，回读一致。没有使用通用 eMMC 包。随后 Recovery 提示 `NEED TO WIPE USERDATA` / `init_user0_failed`；在提前接受清空数据的前提下选择 Factory data reset 后正常初始化。

恢复出厂后工程菜单的 USB Debug、Adb shell 两行消失，但再次读取确认 SKU 仍为 600CN、解锁记录保留，固件仍为 20250825。隐藏原因未定位，不能归因于 SKU 重置或升级。没有 ADB 时，可用 OpenMTP/MTP 取回截图、传文件。

### Magisk 收尾及恢复 ADB

官方 Magisk 30.7 修补本机 boot_a（有 ramdisk，init_boot 为零填充），保留 verity 和强制加密设置，未重签 vbmeta。写入后回读与补丁完全一致。首次 Magisk 要求修复环境，最终在应用内选择“直接安装（推荐）”，不是“安装到未使用的槽位”，完成重启后不再提示。

通过 MTP 传入本地制作的最小 Magisk 模块，在其 service.sh 中执行：

```sh
#!/system/bin/sh
while [ "$(getprop sys.boot_completed)" != "1" ]; do sleep 2; done
sleep 5
setprop persist.duershow.sys.enable.only_self_update false
settings put global development_settings_enabled 1
settings put global adb_enabled 1
setprop persist.sys.usb.config mtp,adb
setprop sys.usb.config mtp,adb
```

作为一组在本机有效，未做逐项必要性测试。模块持续启用 USB 调试，使用者应了解其作用。模块安装重启后 USB 授权弹窗恢复。

### 最终验证与备份注意

Magisk 授权 Shell 后 `su -c id` 返回 `uid=0(root)`；`ro.boot.flash.locked=0`、`ro.boot.vbmeta.device_state=unlocked`，安装限制属性为 false。随后实际通过应用宝安装抖音成功，确认手机本地安装可用。本机没有修改 build.display.id 的 R 后缀。

spd_dump 会裁剪部分签名镜像尾部，部分 NV 导出省略 512 字节头；文件小于分区不一定是不完整，应结合实现和日志核对。`r all` 不包含个人数据，本次也未做整机恢复演练。请勿公开原机 NV、解锁记录、序列号或个人数据。

再次感谢项目、Magisk 和 OpenMTP 的维护者。