TomKing062/CVE-2022-38694_unlock_bootloader #59: Can't exit download mode (autodloader) on Tecno kg5n

WildPowerHammer, 2024-02-17:
I'm trying unlock my bootloader on Tecno from ubuntu, i followed commands in .bat file and now i just tring to exit download mode.

When i trying to execute last command from .bat file which as i understand return original splloader i get error:

fopen(load) failed

full output from 
spd_dump exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec write_part splloader spl.bin write_part uboot_a uboot.bin write_part uboot_b uboot.bin reset

der spl.bin write_part uboot_a uboot.bin write_part uboot_b uboot.bin reset
current exec_addr is 0x65015f08
Waiting for connection (30s)
libusb_control_transfer ok
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND fdl1-dl.bin to 0x65000800
SEND custom_exec_no_verify_65015f08.bin to 0x65015f08
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
KEEP_CHARGE FDL1
SEND fdl2-dl.bin to 0x9efffe00
FDL2: incompatible partition
EXEC FDL2
fopen(load) failed

So, now i just want to unbrick my phone if this possible

WildPowerHammer, 2024-02-17:
I looked at #58 but i'm just following commands from latest .bat (at least i think its latest)

TomKing062, 2024-02-18:
#47

WildPowerHammer, 2024-02-18:
When i try to execute last command i get this:

current exec_addr is 0x65015f08
Waiting for connection (30s)
libusb_control_transfer ok
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND fdl1-dl.bin to 0x65000800
SEND custom_exec_no_verify_65015f08.bin to 0x65015f08
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
KEEP_CHARGE FDL1
SEND fdl2-dl.bin to 0x9efffe00
FDL2: incompatible partition
EXEC FDL2
fopen(load) failed

screenshot for for better understanding
[image]

TomKing062, 2024-02-18:
use u-boot-spl as spl.bin, and (if needed) use fdl2-sign/u-boot-sign as uboot_a/uboot_b

WildPowerHammer, 2024-02-19:
Technically I got out of boot mode and even unlocked bootloader, but now the phone is in bootloop and I don't know if there is anything I can do about it. If you know what else I can do I would be grateful.
[image]

At least it was interesting, and maybe someday I'll understand what I was doing before I did it.
Thank you anyway

TomKing062, 2024-02-20:
userdata need to be format, flash pac firmware should work

WildPowerHammer, 2024-02-20:
I am unable to access fastboot, recovery or any other function on my phone. Therefore, I cannot format userdata or flash anything in this state. If you have any suggestions for breaking the bootloop, please let me know.

TomKing062, 2024-02-20:
pac firmware is for u2s mode

TomKing062, 2024-02-23:
lost splloader backup (including autodloader) has a workaround in batch/shell now

closed 2024-02-23 by TomKing062