TomKing062/CVE-2022-38694_unlock_bootloader #61: hisense a7cc pro

elonuziel, 2024-02-18:
Hello there! I'm having some issues with rooting my phone. I came across a Reddit post (https://www.reddit.com/r/eink/comments/15vqik0/hisense_a7cc_full_google_rooting_guide/) that provided some guidance, but it wasn't until I read https://www.discuss.com.hk/viewthread.php?tid=31297072 and https://codeniz.com/2023/10/07/hisense-a5-pro-root-%E8%88%87-google-%E6%9C%8D%E5%8B%99/ (which had helpful pictures) that I realized my phone already had an unlocked bootloader from the store where I bought it. I was thrilled, but unfortunately, I had already soft-bricked my phone by running `adb reboot autodloader`. 

I've tried various methods to restore my phone, but nothing seems to be working. When I attempted `spd_dump exec_addr 0x3f28 fdl fdl1-dl.bin 0x3f28 fdl uboot-mod.bin 0x9efffe00 exec read_part boot 0 35M boot.img reset`, the output was `custom_exec_no_verify_3f28.bin does not exist`. I've tried searching online for a solution, but I'm a beginner and couldn't find anything that I understand.

At this point, I just want to get my phone working again without any changes, since it was functioning correctly before. However, if possible, I'd also like to have Magisk installed. I thought it would be cool to have root access, but it's not necessary.

I would really appreciate some help. Thank you!

TomKing062, 2024-02-20:
still #47
```
spd_dump fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec write_part splloader u-boot-spl-16k-sign.bin reset
```

elonuziel, 2024-02-20:
thank you so much! able to boot normally!!!

 also with some help was able to install magisk :)

dragen1860, 2024-06-28:
@TomKing062  same error. I need to download `custom_exec_no_verify_3f28.bin`, could you kindly provide a link? thnak you.