TomKing062/CVE-2022-38694_unlock_bootloader #62: Confused with the unlock wiki page

nairpaa, 2024-02-22:
Hello I want to unlock bootloader on Realme C51, but I am confused.

I followed the [wiki](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/wiki/Unlock#ubuntu) for Ubuntu. I already have the binary `chsize` and `spd_dump`.

[image]

Next there is the following command:

> Copy chsize and spd_dump to tool directory, enter download mode, execute command.

There are a few things that I am confused here:
1. What does `tool directory` mean? Is it just an empty directory containing the `chsize` and `spd_dump` binaries?
2. On the [SupportList](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/wiki/SupportList) page, Realme C51 (and other Realme) has no information about `Download Mode`. In the [issues](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/issues/49) section I also checked, and you answered `Download Mode` is SPRD U2S DIAG mode. So how to access SPRD U2S DIAG mode on realme on Ubuntu Linux?
3. What does `execute command` mean? Does this mean binary `chsize` or `spd_dump`? Are there certain parameters or not?

Anyone please help to explain this. If I finally manage to unlock the bootloader, I will contribute to improving the Unlock wiki page.

TomKing062, 2024-02-22:
device specific files in release

TomKing062, 2024-02-23:
wiki updated, about U2S, it is "power + vol up" / "power + vol down" (most common) / "power + vol up + vol down", if all not work, then have to find testpoint on motherboard.

closed 2024-02-23 by TomKing062