TomKing062/CVE-2022-38694_unlock_bootloader #65: Private Key

gian1245, 2024-03-02:
Hi, I have unlocked your my device(TECNO Spark 8C) using your method. Thank you so much for that.

However, I'm only limited so far in installing GSI Roms. I can't install custom recovery or root since flashing img on this device requires the img to be signed first using the private key. 

Is there anyway that your tool/exploit can obtain the private key used by the OEM. Thank you once again.

gian1245, 2024-03-02:
Update: I was able to patch and sign the magisk patched boot.img and was able to flash it successfully. However, when I restart my system, I get a "Android is corrupt" error and it will not boot at all to the OS. Flashing the stock boot.img fixes this. Any tips on how to solve this?

gian1245, 2024-03-02:
Update: I was able to patch and sign magisk patched boot.img and flash it to my system. However, when I reboot, I will get an "Android is corrupt" error. And it will give me two options which is Try Again or Factory Reset. Neither works. Flashing the stock boot.img fixes this. Any tips on how to solve this?

gian1245, 2024-03-05:
I've been trying to root this phone without success. Can you patched and signed my boot img for me? I have attached my boot.img. Thank you so much!
[boot.img.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/14496112/boot.img.zip)

TomKing062, 2024-03-06:
have no idea about magisk
if we can get keys, then it's "public key" not "private key"

gian1245, 2024-03-06:
It didn't work at first. Then, I think maybe it was not working(I still got "Android is corrupt" after flashing it) because I'm currently running a GSI ROM(Evolution X Android 12) . So what I did was open the flash tool, load up my firmware, then changed the stock boot.img with the that you gave me. It worked perfectly! However, I tried flashing gsi rom again through fastboot with this command only

"fastboot flash system "gsi img"

After flashing, I did a factory reset through recovery and restarted the device. And it ended up getting stuck at the logo only. So question, how can I install magisk with the gsi rom? Thank you so much for the help so far.

gian1245, 2024-03-06:
UPDATE: Changing GSI roms did the trick, to be specific, Lineage OS 21. Thank you so much for your help!!

closed 2024-03-06 by gian1245