TomKing062/CVE-2022-38694_unlock_bootloader #67: please add support for zte blade a54

atesin, 2024-03-02:
zte blade a54 has an unisoc SC9863A cpu, AFAIK was shipped about nov 2023 with android 13

i saw this phone and like it because has 4gb ram and simultaneous dual sim + microsd, also has a big battery and is a known brand... i bought this phone new so cheap in a retail store (in chile), discounted with my credit card, about a price of a senior basic phone!

i want to replace my old phone (umidigi a5 pro) because a touchscren input lag, by migrating all my apps WITH DATA and then remove all traces of google apps (and maybe install microG, i don't care google play since i use aurora store)

for this i need to unlock bootloader and root (to use titanium backup), then remove all google system apks, then enable signature spoofing some way to install microG

how can i do all this?, what is the first step?, will you support this phone?

atesin, 2024-03-05:
as zte blade a54 has the same cpu as blade a5 (unisoc sc9863a), could this procedure linked below help to unlock the blade a54 too?

#4

TomKing062, 2024-03-06:
i only do modification if related files are provided
if can't find official files,there are methods to read files,like use other paid tools or use FDLs of other zte SC9863A(NOTE: this is dangerous and needs knowledge of using 'spd_dump', i will not write down full process)
LAST：NEVER execute 'adb reboot autodloader'

atesin, 2024-03-18:
update: recently, a notification pops up in my phone saying a firmware update is available, i hadn't accept yet... can something be done with this?, like "intercept" the firmware update or something?, maybe an app that "reads" the url and download the firmware file before or while it gets updated?

TomKing062, 2024-03-18:
i notice it is android 13, maybe you can try `fastboot oem pull` with https://github.com/TomKing062/python-fastboot-oem-pull

TomKing062, 2024-03-18:
i only see tecno block custom unlock
zte's update should be safe i think

atesin, 2024-03-19:
> i hadn't accept yet

too late, it got updated automatically despite my skips :/ (i feel a bit angry, i have no control of my own phone really)

> maybe you can try fastboot oem pull

what is that?, i never heard of it.... on the other hand, i read that partitions are encrypted, and the key is unknown (except for zte/wom xD )

(wom is the chilean carrier that oem'ed the phone, they brought it absurdly cheap)
https://zte.cl/producto/a54-dark-grey/

TomKing062, 2024-08-15:
use sc9863a_zte_blade_a52.zip(for 32bit system) and sc9863a_zte_voyage_40se_v4.zip(for 64bit system)

closed 2024-08-15 by TomKing062