TomKing062/CVE-2022-38694_unlock_bootloader #69: ZTE Blade A51 bootloop

Javialonqv, 2024-03-09:
Ok, so, after unlocking the bootloader for ZTE Blade A51, it's stuck in a bootloop, and that's because I forgot to wipe all the data, and the command to fix it doesn't work for me, first it gives me an error "custom_exec_no_verify_0.bin does not exist." The file I had is "custom_exec_no_verify_4ee8.bin", I tried changing the name and now it says
"sc misc-fastbootd.bin reset
current exec_addr is 0x0
bad command args".

Please, I need help with this.

Javialonqv, 2024-03-09:
**UPDATE**

Repeating the unlock process allows me to boot into the recovery, but make a factory reset don't solve the issue, so, I'm stuck again.

TomKing062, 2024-03-09:
use full pac firmware, google "ZTE Blade A51 firmware"

Javialonqv, 2024-03-09:
The only way I've seen to turn it off is to run the script again so I can get into fastbootd (using the key combination didn't work) then from there I can hit the shutdown button, but if I have the device connected the device restarts again.

Javialonqv, 2024-03-09:
**UPDATE**

I was able to flash the stock firmware starting the process **EXACTLY** when the device restarts, after the flash, the device now works fine.

closed 2024-03-09 by Javialonqv