TomKing062/CVE-2022-38694_unlock_bootloader #72: slow spd_dump download speed on Linux, libusb

RadGoodNowYT, 2024-03-13:
LOG
> root@linuxmint-rad:/home/rad/spddump/spreadtrum_flash# ./spd_dump exec_addr 0x65015f48 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec write_part boot_a magisk_patched-26404_nbwmo.img reset
> current exec_addr is 0x65015f48
> Waiting for connection (30s)
> libusb_control_transfer ok
> CHECK_BAUD bootrom
> BSL_REP_VER: "SPRD3\0"
> CMD_CONNECT bootrom
> SEND fdl1-dl.bin to 0x65000800
> SEND custom_exec_no_verify_65015f48.bin to 0x65015f48
> EXEC FDL1
> CHECK_BAUD FDL1
> BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
> CMD_CONNECT FDL1
> KEEP_CHARGE FDL1
> SEND fdl2-dl.bin to 0x9efffe00
> FDL2: incompatible partition
> EXEC FDL2
> DISABLE_TRANSCODE
> ENABLE_WRITE_RAW_DATA
> Answer "yes" to confirm the "write partition" command: yes
> file size : 0x4000000
> [=                                       ] 3.1%
> [==                                      ] 6.2%
> [===                                     ] 9.4%
> [=====                                   ] 12.5%
> [======                                  ] 15.6%
> [=======                                 ] 18.8%
> [========                                ] 21.9%
> [==========                              ] 25.0%
> [===========                             ] 28.1%
> [============                            ] 31.2%
> [=============                           ] 34.4%
> [===============                         ] 37.5%
> [================                        ] 40.6%
> [=================                       ] 43.8%
> [==================                      ] 46.9%
> [====================                    ] 50.0%
> [=====================                   ] 53.1%
> [======================                  ] 56.2%
> [=======================                 ] 59.4%
> [=========================               ] 62.5%
> [==========================              ] 65.6%
> [===========================             ] 68.8%
> [============================            ] 71.9%
> [==============================          ] 75.0%
> [===============================         ] 78.1%
> [================================        ] 81.2%
> [=================================       ] 84.4%
> [===================================     ] 87.5%
> [====================================    ] 90.6%
> [=====================================   ] 93.8%
> [======================================  ] 96.9%
> [========================================] 100.0%
> load_partition: boot_a, target: 0x4000000, written: 0x4000000
> root@linuxmint-rad:/home/rad/spddump/spreadtrum_flash# 
>
I've been waiting for 3-6 minutes to write 64M boot. But in Windows, spd_dump will write the boot in 5 seconds. Is there any way to fix the speed or I need to modify code of the program?
Distro:Linux Mint 21.3 x86_64

TomKing062, 2024-03-14:
set blk_size
main(dev) branch has large default blk_size, stable branch has small default blk_size
a lagrer default blk_size will cause some issues, so i created stable branch

TomKing062, 2024-03-16:
i tested on libusb win, dev branch is slow,stable is normal (when set blk_size 0xff00 on stable)

i will check if there is any method to increase cache size, or i will let libusb always use old protocol instaed of raw data protocol in dev branch

TomKing062, 2024-03-16:
disabled raw data protocol for libusb

closed 2024-03-16 by TomKing062