TomKing062/CVE-2022-38694_unlock_bootloader #73: Can't flash vbmeta on windows10,ud710

chw2333, 2024-03-15:
lts log
`C:\Users\Administrator\Desktop>spd_dump fdl fdl1.bin 0x5500 fdl fdl2.bin 0x9efffe00 exec write_part vbmeta vbmeta.img reset
Waiting for connection (30s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND fdl1.bin to 0x5500
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
KEEP_CHARGE FDL1
SEND fdl2.bin to 0x9efffe00
FDL2: incompatible partition
EXEC FDL2
DISABLE_TRANSCODE
file size : 0x100000
Answer "yes" to confirm the "write partition" command: yes
[==                                      ] 5.9%
[====                                    ] 11.7%
[=======                                 ] 17.6%
[=========                               ] 23.4%
[===========                             ] 29.3%
[==============                          ] 35.2%
[================                        ] 41.0%
[==================                      ] 46.9%
[=====================                   ] 52.7%
[=======================                 ] 58.6%
[=========================               ] 64.5%
[============================            ] 70.3%
[==============================          ] 76.2%
[================================        ] 82.0%
[===================================     ] 87.9%
[=====================================   ] 93.8%
[======================================= ] 99.6%
timeout reached`
Only vbmeta cannot be flashed.
Windows version:19045.3930

KawaiiSparkle, 2024-03-16:
你有没有试过刷vbmeta_bak分区 | Have you ever tried flashing the vbmeta_bak partition

KawaiiSparkle, 2024-03-16:
BYD原来是群友啊（）

TomKing062, 2024-03-16:
what fdl you use, since stock uboot will check signature of vbmeta

chw2333, 2024-03-16:
感谢大佬的帮助，问题已经解决了
刷回vbmeta_bak然后清除vbmeta就好了

closed 2024-03-16 by TomKing062