TomKing062/CVE-2022-38694_unlock_bootloader #78: Bootloader unlocked, but, how can I root with Magisk?

Javialonqv, 2024-03-25:
I saw the guide on how to root Android (11+) with Magisk in Systemless mode, but before doing anything, I have some doubts that I would like to resolve.

1.- How do I extract or obtain the **vbmeta.bin** and **rsa4096_vbmeta.pem** files?
2.- When I unpack the **magisk_patched-*.img** file, what should I do? Because I don't see much point in unpacking it and packing it again.
3.- In case a bootloop occurs, I will have to use **_fastboot flash vbmeta_**, but with which file?

Sorry if these doubts seem a bit absurd, but I really don't have much experience flashing and rooting Androids.

Javialonqv, 2024-03-25:
From what I understand, I need to sign the magisk_patched-*.img file and the vbmeta.img file so that the device does not go into a bootloop, am I right?

TomKing062, 2024-03-26:
unpack and repack can be skip, sign boot with rsa4096_any.pem
vbmeta doesn't need any modify, see #77
keys, [here](https://github.com/TomKing062/vendor_sprd_proprietories-source_packimage/tree/main/sign_image/config-unisoc)

Javialonqv, 2024-03-26:
Ok, so, I just extract the boot.img, patch it with Magisk, sign it with rsa4096 file, and flash it, that's it? But, in the case I get into a bootloop, the guide says something of flashing the vbmeta, the file it's inside of the PAC file? Like the boot.img? Or where I can find it?

Javialonqv, 2024-03-27:
**UPDATE:**

I tried to root the device flashing the signed boot and all the vbmeta partitions (in a/b).
I unpacked and repacked the magisk_patched.img file and sign it with avbtool with the rsa4096_boot.pem file,
but it stills getting in a bootloop.
How can I fix it?

Javialonqv, 2024-03-30:
So, if I use `fastboot flash --disable-verity vbmeta vbmeta.img` with my stock vbmeta.img file, it should disable AVB on my ZTE Blade A51 (sc9863a), or it will be stuck at the bootlogo (bootloop)? Because I cant try it now by my own.

Javialonqv, 2024-04-01:
**UPDATE:**

I tried flashing the stock vbmeta with `fastboot --disable-verity flash vbmeta vbmeta.img` and the device boot, however, after flashing the patched boot, the devices stills getting in a bootloop, nothing changed.

TomKing062, 2024-04-04:
try change between slot a and b, and may still need to wipe data

Javialonqv, 2024-04-05:
**UPDATE:**

I was able to root the device, however, I did a lot of things, and I'm not really sure what made it work, so this is what I did:

1.- Flash the stock vbmeta file with the disable verity flag in the a partition.
`fastboot --disable-verity flash vbmeta_a vbmeta.img`
2.- WITHOUT rebooting, enter to recovery and wipe all user data.
3.- Reboot and then enter on fastboot again.
4.- Instead of sign the patched-boot.img with the command on the wiki
`python2 avbtool add_hash_footer --image AIK/image-new.img --partition_name boot --partition_size 67108864 --key rsa4096_boot.pem --algorithm SHA256_RSA4096 --prop com.android.build.boot.fingerprint:$FINGERPRINT --prop com.android.build.boot.os_version:$OS_VERSION --salt 7A91E47F8D2CFB95DCCFF13305EE3F07EDCF83A42660A811F3724E1E8B463284`
replace `$FINGERPRINT` with the fingerprint of your stock boot.img (you can do this by using `python avbtool info_image --image boot.img`) and `$OS_VERSION` with `11`. Then add the following flag `--rollback_index 1`.
5.- Sign the patched-boot.img with that command.
6.- Flash it to the device at the a partition with `fastboot flash boot_a patched-boot-signed.img`.
7.- Enter recovery and wipe userdata again.
8.- Reboot and install the magisk app!

closed 2024-04-05 by Javialonqv

zoomver, 2024-04-05:
Hello.
Please write detailed instructions on how you managed to install root. I would really like to try this on the Zte blade A31.
Thank you.

Javialonqv, 2024-04-05:
I don't know if this could work for ZTE Blade A31, but here is it (Assuming you already have an unlocked bootloader).

**Get the STOCK boot.img and vbmeta-sign.img info**
1.- Download [SPD Upgrade Tool](https://spdflashtool.com/) and your stock firmware (.pac file).
2.- Open the .pac file and DON'T CLOSE THE PROGRAM, the extracted files will be at a temp folder called "ImageFiles", extract the boot and vbmeta-sign files.
3.- Using avbtool, execute the following command: `python avbtool info_image --image <path_to_your_stock_boot.img>`
4.- You will have an output with info about the file, save the following information (you will need it):
`Rollback Index` `Prop: com.android.build.boot.fingerprint` and `Prop: com.android.build.boot.os_version`.

**Patch the stock boot.img with Magisk**
1.- Install the [Magisk](https://github.com/topjohnwu/Magisk/releases) app in your phone.
2.- Transfer your stock boot.img to your phone.
3.- Patch it using the Magisk app.
4.- Transfer the output patched boot to your computer again.

**Sign the patched boot with avbtool**
As I see that your device runs Android 11, run the following command:
`python2 avbtool add_hash_footer --image AIK/image-new.img --partition_name boot --partition_size 67108864 --key rsa4096_boot.pem --algorithm SHA256_RSA4096 --prop com.android.build.boot.fingerprint:$FINGERPRINT --prop com.android.build.boot.os_version:$OS_VERSION --rollback_index <index> --salt 7A91E47F8D2CFB95DCCFF13305EE3F07EDCF83A42660A811F3724E1E8B463284`
replacing `--rollback_index <index>` with your info, same with the `$FINGERPRINT` and `$OS_VERSION`.

**Flash the stock vbmeta and patched-boot-signed.img**
1.- Enter your device into fastboot.
2.- Flash your stock vbmeta image with `fastboot --disable-verity flash vbmeta_a <path_to_your_stock_vbmeta>`
3.- Flash the patched-boot-signed.img with `fastboot flash boot_a <path_to_your_patched_boot>`
4.- WITHOUT RESTARTING THE DEVICE, enter into recovery and wipe user data, otherwise it will enter in a bootloop.
5.- Restart.

I am ready to help you if needed.

Javialonqv, 2024-04-05:
There's something that doesn't make any sense, when flashing stock vbmeta with `--disable-verity` flag, we are disabling **AVB (Android Verified Boot)**, and in theory, we **DON'T** need to sign the patched boot with the key, so, that's a contradiction that I don't understand, maybe the patched boot signed can work without flashing the stock vbmeta with that flag, just if anyone want to try.

zoomver, 2024-04-06:
Hello.
Thank you very much for the instructions.
I will experiment.

Javialonqv, 2024-04-06:
> Hello. Thank you very much for the instructions. I will experiment.

You're welcome, and I just tried, and there's no need to flash the stock vbmeta with the specified flag, just flash the patched and signed boot image, wipe all data and that's it.

Javialonqv, 2024-04-06:
> In theory, we **DON'T** need to sign the patched boot with the key, so, that's a contradiction that I don't understand, maybe the patched boot signed can work without flashing the stock vbmeta with that flag, just if anyone want to try.

After doing some testing with `adb shell getprop`, I see that for some reason, **AVB** is not disabled even after flashing vbmeta with `--disable-verity` (at least on my device), so practically, flashing vbmeta is a completely useless step.

zoomver, 2024-04-07:
Hello.
Please tell me, I'm trying to sign boot.img.
But avbtool throws an error when specifying rollback_index.
Thank you if you can help.

zoomver, 2024-04-07:
Hello.
Thank you very much.
It's really great.
I did it all right.
Thank you for the help and the hint.
Good luck and good health to you.
👍👍👍👍😄👍👍👍👍

Skorpion96, 2024-05-20:
> **UPDATE:**
>
> I was able to root the device, however, I did a lot of things, and I'm not really sure what made it work, so this is what I did:
>
> 1.- Flash the stock vbmeta file with the disable verity [...]

> I don't know if this could work for ZTE Blade A31, but here is it (Assuming you already have an unlocked bootloader).
>
> **Get the STOCK boot.img and vbmeta-sign.img info** 1.- Download [SPD Upgrade [...]

Worked on ZTE Blade A52
[Root ZTE Blade A52.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/15381198/Root.ZTE.Blade.A52.zip)

Javialonqv, 2024-05-30:
> Worked on ZTE Blade A52
> [Root ZTE Blade A52.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/15381198/Root.ZTE.Blade.A52.zip)

Yes, that's because the new models of ZTE Blade such as A51 or A52 seems to be using a public RSA key, instead of the old models, that use a private one.