TomKing062/CVE-2022-38694_unlock_bootloader #80: ZTE Blade A31 (SC9863A chip) Possible causes of bootloop?

zoomver, 2024-03-27:
Hello!
Please tell me:
after installing boot.img patched using magisk according to all your instructions, bootloop occurs.
Could the following indicators obtained using adb shell => getprop be the reason for this:
getprop
ro.boot.dm verity_disabled:0
getprop
ro.boot.force_normal_boot:1
getprop
ro.boot.vbmeta.digest: (the digest is listed here)
getprop
ro.boot.vb meta.invalidate_on_error:yes
getprop
ro.boot.veritymode:enforcing

If the specified indicators can be the cause of bootloop, what method can I use to change them?

TomKing062, 2024-03-30:
fastboot --disable-verity flash vbmeta vbmeta.img
this disable dm-verity
but i don't think this affect boot check process

zoomver, 2024-03-31:
Hello.
Thank you for your reply.
Please tell me, and the other parameters do not affect the occurrence of bootloop?

TomKing062, 2024-04-04:
try change between slot a and b, and may still need to wipe data

zoomver, 2024-04-04:
Hello.
Okay, I get you.
I will experiment.
thanks.

zoomver, 2024-04-07:
Hello!
Thank you very much for your work. It really works.
I managed to install root access on my ZTE blade A31.
Thank you very much and good luck in your life and work.

closed 2024-04-07 by zoomver