TomKing062/CVE-2022-38694_unlock_bootloader #83: Getting error unlocking the bootloader on itel p40+

infinixs5lite, 2024-03-31:
Getting error unlocking the bootloader on itel p40+ 
[image]
[image]
[image]

infinixs5lite, 2024-03-31:
Is it am not on download mode?

infinixs5lite, 2024-03-31:
Do I have to use test point?

TomKing062, 2024-03-31:
seems a extre check in uboot, update later(maybe tomorrow)

infinixs5lite, 2024-03-31:
> seems a extre check in uboot, update later(maybe tomorrow)

Pls try because I'm about to use anger and sell the device, because is my first time using unisoc

TomKing062, 2024-04-01:
uploaded

infinixs5lite, 2024-04-01:
My phone got hard brick after typing yes

TomKing062, 2024-04-02:
> file you send is 230619, be careful with any update

the script/batch will recover partition to dumped backup, if dumped backup is destroyed, 230619 will be write to device
the only possible brick is device already newer than 230619

infinixs5lite, 2024-04-02:
I don't understand

TomKing062, 2024-04-02:
for windows
```
set /p StorageType=<StorageType
set spl_bak=u-boot-spl-16k-%StorageType%-sign.bin
set uboot_a_bak=fdl2-sign.bin
set uboot_b_bak=fdl2-sign.bin
spd_dump exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec write_part splloader %spl_bak% write_part uboot_a %uboot_a_bak% write_part uboot_b %uboot_b_bak% reset
```

infinixs5lite, 2024-04-02:
Phone not detected anymore

TomKing062, 2024-04-04:
after flash u-boot-spl-16k-%StorageType%-sign.bin and fdl2-sign.bin ?
or not flashable

use key to re-enter u2s, it's normal that screen keeps unchanged

closed 2024-08-15 by TomKing062