TomKing062/CVE-2022-38694_unlock_bootloader #90: Support for Motorola Moto E13 (Codename SABAHL)

RhyanDumer456, 2024-04-13:
I have always unlocked the bootloader of my cell phones since the Moto E LTE 2015, but this Moto E13 2023, which is Unisoc T606, i cannot unlock using official methods and Motorola claim that Android Go devices cannot be unlocked, so I would be very happy if you added support for this device. I will send the files that I believe are necessary. If anything else is needed, I'm willing to help.
[Moto e13 sabahl.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/files/14965082/Moto.e13.sabahl.zip)

TomKing062, 2024-08-03:
uploaded to release

closed 2024-08-15 by TomKing062

TomKing062, 2025-12-14:
fastboot oem/flashing lock_bootloader

TomKing062, 2025-12-14:
https://github.com/TomKing062/spreadtrum_flash/blob/d24c21a0c93a545c0130668caa1826d59ceebe48/README.md?plain=1#L149