TomKing062/CVE-2022-38694_unlock_bootloader #93: Hisense A7 rooted, but gets into bootloop every time it auto shut down after battery drained.

daopeng, 2024-04-30:
My Hisense A7 phone has been rooted and all, but it won't start after auto shutting down when battery is drained. It just gets into a bootloop or gets stuck at the startup logo. I have encountered this several times, and each time I have to do it all over again. I don't know if anyone ever had the same situation, or what caused this, and is there any way to prevent this? Many thanks.

TomKing062, 2024-05-01:
hisense part, or some magisk modules part i think
"do it all over again" is just erase or full flash?

closed 2024-05-01 by TomKing062

daopeng, 2024-05-03:
Sorry for the late reply. Yeah, my first thought is some magisk module, too, so after twice of this happening, I installed an anti bootloop module, but it's not working for me, I still had to go through a system restore by TF card, patching the inage and flashing it back in. The good thing is this phone is only for reading, so there isn't many apps or personal data on it. But still it's a pain in the ass every time I don't charge in time. I just wish maybe someone had some knowledge about this and knew how to fix it. Thanks a lot for replying.