TomKing062/CVE-2022-38694_unlock_bootloader #94: 希望加入对 Nokia T20 (TA-1392/TA-1394/TA-1397) 的支持

HikariCalyx, 2024-06-07:
对应的 fdl、uboot、splloader 已随附件给出。
此机型搭载 T610 (ums512) 处理器。
[RDD_bootloader.zip](https://github.com/user-attachments/files/15737853/RDD_bootloader.zip)

TomKing062, 2024-08-03:
a secboot check here
not sure if this differ between develop/consumer version
will upload to release later

TomKing062, 2024-08-03:
[ums512_Nokia_T20_android11.zip](https://github.com/user-attachments/files/16479063/ums512_Nokia_T20_android11.zip)

a11 doesn't have secboot and download auth check

closed 2024-08-16 by HikariCalyx

TomKing062, 2025-04-01:
a13 won't be supported

Noverisp3, 2025-05-30:
hey, i found this in spd_dump

[image]

disable verity? is it usable for turn off AVB? and i have this log

FDL2 >verity 0
Warning: disable dm-verity needs a write-verification-disabled FDL2
unexpected response (0x0084)

What does this "disable dm-verity needs a write-verification-disabled FDL2" mean? and how i disable it?