TomKing062/CVE-2022-38694_unlock_bootloader #95: Support for Anbernic T820 devices (ums9620_2h10)

TheGammaSqueeze, 2024-06-15:
Hello there, thank you for the great work that you have done! 

I am the developer of a custom firmware for Anbernic handheld gaming devices. My previous work was with their T618 chips. 
They have now released devices with T820 chips, which of course can't be bootloader unlocked using the old method. 

If I gave you the files that I've dumped from the new devices, would you be able to help me figure how to unlock these devices? 
I am happy to fund a device for you for development and testing purposes :)

I also have the full .pac files if you need these, but here are the uboot and spl files from two of their devices:

[RG556.zip](https://github.com/user-attachments/files/15853477/RG556.zip)
[RG-CUBE.zip](https://github.com/user-attachments/files/15853487/RG-CUBE.zip)

Please do let me know if you are able to help, thank you.

TheGammaSqueeze, 2024-06-15:
[RG556-ums9620_coolpad_GoldCentury_y60.txt](https://github.com/user-attachments/files/15854971/RG556-ums9620_coolpad_GoldCentury_y60.txt)

It's probably a bad idea to try another device package on this one, but I can confirm everything works except when sending spl-emmc-unlock.bin. I suspect we would need to patch the spl with the ones I provided from the stock firmware?

TomKing062, 2024-06-20:
[RGs.zip](https://github.com/user-attachments/files/15909575/RGs.zip)

fdl1-dl/fdl2-dl from coolpad y60 can be reused

TheGammaSqueeze, 2024-06-20:
Thank you for this!
I replaced the files used on coolpad with the patched ones you provided.

I am trying this on RG-Cube, but it seems that it fails to send spl-emmc-unlock.bin.

[cube_fail.txt](https://github.com/user-attachments/files/15910675/cube_fail.txt)

Any idea on this?

TheGammaSqueeze, 2024-06-24:
Hi @TomKing062, Anbernic are prepared to send you devices to help us unlock the bootloader. Are you interested in this?

TomKing062, 2024-06-25:
didn't find error in modification
(just curious)Anbernic has vbmeta key for fastboot method, or can change little kernel code and compile a specific lk to unlock

TheGammaSqueeze, 2024-06-25:
USB send fail according to logs. Bootloader is still locked.

If this helps, here are PAC files for the devices

556: https://drive.google.com/file/d/1hYuUPBTHAghZXXVt5urnAhJP-_vofg3K/view?usp=sharing
Cube: https://drive.google.com/file/d/1DBl9ZwvyqUdjMmetA2ekYWKS1uUcmI0b/view?usp=sharing

If you want me to send you devices at no cost to you, please let me know and I can arrange this.

TomKing062, 2024-06-25:
maybe 820 need a different custom_exec file
[custom_exec_no_verify_65012f48.zip](https://github.com/user-attachments/files/15968486/custom_exec_no_verify_65012f48.zip)

TheGammaSqueeze, 2024-06-25:
Result is still the same:

C:\cube\ums9620_coolpad_GoldCentury_y60_v2>spd_dump exec_addr 0x65012f48 fdl spl-emmc-unlock.bin 0x65000800
branch:stable, sha1:457e3ec04db19fdcafac83007622cd4f00932b51
current exec_addr is 0x65012f48
Waiting for connection (30s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND spl-emmc-unlock.bin to 0x65000800
SEND custom_exec_no_verify_65012f48.bin to 0x65012f48
EXEC FDL1
**usb_send failed (0 / 1)**

C:\cube\ums9620_coolpad_GoldCentury_y60_v2>spd_dump exec_addr 0x65012f48 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0xb4fffe00 exec verbose 2 read_part miscdata 8192 64 m.bin reset
branch:stable, sha1:457e3ec04db19fdcafac83007622cd4f00932b51
current exec_addr is 0x65012f48
Waiting for connection (30s)
**usb_send failed (0 / 1)**

TheGammaSqueeze, 2024-06-25:
The custom_exec_no_verify_65012f48.bin works fine for all other activities, only when we send spl-emmc-unlock.bin as fdl it fails.

TheGammaSqueeze, 2024-06-25:
> didn't find error in modification (just curious)Anbernic has vbmeta key for fastboot method, or can change little kernel code and compile a specific lk to unlock

They cannot share any keys with us unfortunately.

TheGammaSqueeze, 2024-06-25:
Could there be a possibility that FDL1 method will work instead of SPL method?
I also have access to some sources for ums9620 (they are a little older than this T820 chip) if that helps:

[image]

Its not related to the Anbernic device, but might help? I can share privately.

TomKing062, 2024-06-26:
wow fdl way works
[image]

TheGammaSqueeze, 2024-06-26:
Would you be able to help me find the correct offsets for sml_a, teecfg_a, trustos_a for my devices? Sorry I am not familiar with Ghidra/IDA 

[image]

TomKing062, 2024-06-26:
[RG-fdl.zip](https://github.com/user-attachments/files/15982788/RG-fdl.zip)

TheGammaSqueeze, 2024-06-26:
Success!

[image]

[image]

Thank you very much, I will send some money via Paypal as a thank you :)

closed 2024-06-26 by TheGammaSqueeze

DRCRecoveryData, 2024-10-19:
> ## [RG-fdl.zip](https://github.com/user-attachments/files/15982788/RG-fdl.zip)
> 
> actually fdl way doesn't need to erase splloader

Hi, does it support T157 unisoc devices?