TomKing062/CVE-2022-38694_unlock_bootloader #97: Signature Key for WHOOP TAB-8US2?

gittyaiden, 2024-07-06:
Trying to unlock the bootloader for this device. Do you have a signature key that works for this?

gittyaiden, 2024-07-07:
[image]

zoomver, 2024-07-08:
Hello.
Unfortunately, you will not be able to unlock the bootloader using this method.
At the moment, all manufacturers use their own keys.
This method was used on the Android 9 operating system. And the private key used in this method was leaked to the Internet.

gittyaiden, 2024-07-08:
Is there anything else I can do to unlock the bootloader?

zoomver, 2024-07-09:
Hi gittyaiden.
The only method to unlock the bootloader right now is the method from TomKing062.
You need to ask him to try to write the unlock files. At the same time, you need to provide the following files:
u-boot-spl-16k-sign.bin
Fdl-1.bin
Fdl-2.bin
Specify the type (brand) of the smartphone processor. The Android OS version.
You can find these files in the stock firmware for your gadget. If you don't have these files, try to find them (on forums, websites).
Without these files, TomKing062 will not be able to help you.
Good luck to you.

gittyaiden, 2024-07-22:
> t310?

Yes! I have a T310 chipset

gittyaiden, 2024-07-26:
Was able to unlock the bootloader using the hotavek method. My silly self
was supposed to combine the two segments of my unlock code together to
unlock the bootloader. Now that this has been resolved. I wonder how I’ll
be able to root my device now

RadGoodNow, 2024-07-28:
> Now that this has been resolved. I wonder how I’ll be able to root my device now

you need to patch boot.img with magisk and use [this method](https://www.hovatek.com/forum/thread-32674.html) to sign image. without signing you'll bootloop your device because bootloader won't accept your custom boot.img

gittyaiden, 2024-07-28:
this makes my brain hurt 😭. You got anything super easy to do?

RadGoodNow, 2024-07-28:
> You got anything super easy to do?

Isn't this already super easy? I'll write what steps you will be need to do
1. [Download and install Magisk manager](https://github.com/topjohnwu/Magisk/releases/tag/v27.0) 
2. Extract boot.img from your firmware file or device
3. Open Magisk app, press on first Install button and select "Select and patch a file" option. Choose your stock boot.img file and patch it
4. Extract patched boot file to pc
5. Download and install [Python](https://www.python.org/)

Now all what you need to do now is follow hovatek guide that I've sent
1. Download avbtool.py for your python version(you can find links in the guide)
2. Generate or [download existing key](https://mega.nz/file/tjAg0K4R#sN-XJJe0ca5KaDTkSL_xhEXQ2fxZXXE73SlDNm59q10) for signing 
3. Open terminal and run this command: `python avbtool add_hash_footer --image patched_boot.img --partition_name boot--partition_size 67108864 --key rsa4096_vbmeta.pem --algorithm SHA256_RSA4096`
4. Flash your boot through fastboot or ResearchDownload. Also you can use spd_dump to flash if you know how to use it

gittyaiden, 2024-07-28:
There’s no firmware for my WHOOP device. And yes I tried going to the
manufacturer site to look for firmware, but there was nothing. I even tried
contacting customer support and they couldn’t do anything.

RadGoodNow, 2024-07-28:
> There’s no firmware for my WHOOP device

I think it's still possible to extract your stock boot image from device. Almost all modern unisoc phones have [hidden command](https://github.com/TomKing062/python-fastboot-oem-pull) that can be used to extract partitions and maybe your device has this command too
So you install all required libs through pip and run this command: `python main.py oem pull boot_a`

gittyaiden, 2024-07-28:
Did all of the steps in the hidden command link but it just comes up as
“[Errno 2] No such file or directory”

RadGoodNow, 2024-07-28:
> “[Errno 2] No such file or directory”

clone that repo and run terminal there. should work

gittyaiden, 2024-07-28:
Sorry. I don’t understand what you mean by this