TomKing062/CVE-2022-38694_unlock_bootloader #98: 我正在等待对摩托罗拉 E13 手机的支持

nightstar-su, 2024-07-10:
我正在等待摩托罗拉 E13 4+64 Android 13 Go UNISOC T-606 UMS9230_1h10_go 的设备支持。以下是文件 fdl1.bin, lk-fdl2.bin, u-boot-spl-16k-sign.bin

[fdl1 lkfdl2 ubootspl16ksign.zip](https://github.com/user-attachments/files/16167204/fdl1.lkfdl2.ubootspl16ksign.zip)

TomKing062, 2024-07-22:
try [test.ums9230_universal_unlock.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/releases/download/1.71/test.ums9230_universal_unlock.zip)

gatesahl, 2024-07-29:
> try [test.ums9230_universal_unlock.zip](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/releases/download/1.71/test.ums9230_universal_unlock.zip)

**I'm sorry**, because I haven't been able to test it yet, however, **it's among what I plan** for the short term.

I saved it, but I still ask you not to delete it, I will show it so that people with this SOC can test it. I will **try my best** to come back with feedback.

Finally, I have to say, **congratulations on your contributions, your work is inspiring and revolutionary. I assume that the results of these files this time are also obviously successful. I really appreciate all your work.**

TomKing062, 2024-08-03:
uploaded to release

closed 2024-08-15 by TomKing062