TomKing062/CVE-2022-38694_unlock_bootloader #100: ums9230 Realme RMX3760 bootloader unlock problem

les1er1, 2024-07-19:
I am using android 13 (RMX3760export_13_A.30), debugger and OEM unlocking are enabled. I used the script for RMX3760 and RMX3762 versions, the script itself completes successfully without any critical errors, but the bootloader status does not change and the system just boots. I noticed that on the second step (second spd_dump command) the following bootloader version is displayed:
 BSL_REP_VER: “Spreadtrum Boot Block version 1.1\0{E#R*R!O&R}\0”.
I couldn't find any information about this on the Internet.
Also the last command spd_dump is not executed from the first time, after “EXEC FDL1” a couple of lines “CHECK_BAUD FAIL” and after that “timeout reached” are displayed, and the phone has to be rebooted via POWER and VOL_UP, only after that the last command is executed successfully. What to do?

Here is most of the log:
```
> spd_dump exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec r splloader r uboot_a r uboot_b e splloader e splloader_bak reset

branch:stable, sha1:47055dfdba23f2cd76722d9128b682e3c4c9fc7b
current exec_addr is 0x65015f08
Waiting for connection (30s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND fdl1-dl.bin to 0x65000800
SEND custom_exec_no_verify_65015f08.bin to 0x65015f08
EXEC FDL1
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
KEEP_CHARGE FDL1
SEND fdl2-dl.bin to 0x9efffe00
FDL2: incompatible partition
EXEC FDL2
DISABLE_TRANSCODE
Reading Partition List

dump_partition: user_partition+0x0, target: 0x8000, read: 0x8000

standard gpt table saved to pgpt.bin
skip saving sprd partition list packet
partition list saved to partition.xml
Total number of partitions: 77

dump_partition: splloader+0x0, target: 0x40000, read: 0x40000

dump_partition: uboot_a+0x0, target: 0x300000, read: 0x300000

dump_partition: uboot_b+0x0, target: 0x300000, read: 0x300000
Answer "yes" to confirm the "erase partition" command: yes
Answer "yes" to confirm the "erase partition" command: yes
"don't continue if you see find port failed, just close and re-run this batch"

> pause
Press any key to continue . . .

> spd_dump exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec w uboot_a fdl2-cboot.bin w uboot_b fdl2-cboot.bin reset
branch:stable, sha1:47055dfdba23f2cd76722d9128b682e3c4c9fc7b
current exec_addr is 0x65015f08
Waiting for connection (30s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND fdl1-dl.bin to 0x65000800
SEND custom_exec_no_verify_65015f08.bin to 0x65015f08
EXEC FDL1
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0{E#R*R!O&R}\0"
CMD_CONNECT FDL1
KEEP_CHARGE FDL1
SEND fdl2-dl.bin to 0x9efffe00
FDL2: incompatible partition
EXEC FDL2
DISABLE_TRANSCODE
Answer "yes" to confirm the "write partition" command: yes
file size : 0x18a7d8
load_partition: uboot_a, target: 0x18a7d8, written: 0x18a7d8
Answer "yes" to confirm the "write partition" command: yes
file size : 0x18a7d8
load_partition: uboot_b, target: 0x18a7d8, written: 0x18a7d8

> set StorageType=ufs
> set spl_bak=splloader.bin
> uboot_a_bak=uboot_a.bin
> uboot_b_bak=uboot_b.bin
> chsize splloader.bin 0xfa68
> if 0 EQU 1 (set spl_bak=u-boot-spl-16k-ufs-sign.bin  )
> chsize uboot_a.bin 0x18a7d8
> if 0 EQU 1 (set uboot_a_bak=fdl2-sign.bin  )
> chsize uboot_b.bin 0x18a7d8
> if 0 EQU 1 (set uboot_b_bak=fdl2-sign.bin  )
> TIMEOUT /T 10 /NOBREAK
Waiting for  0 seconds, press CTRL+C to quit ...

> set spl_unlock=spl-ufs-unlock.bin

> spd_dump exec_addr 0x65015f08 fdl spl-ufs-unlock.bin 0x65000800
branch:stable, sha1:47055dfdba23f2cd76722d9128b682e3c4c9fc7b
current exec_addr is 0x65015f08
Waiting for connection (30s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND spl-ufs-unlock.bin to 0x65000800
SEND custom_exec_no_verify_65015f08.bin to 0x65015f08
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FAIL
timeout reached

> spd_dump exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec verbose 2 read_part miscdata 8192 64 m.bin reset
branch:stable, sha1:47055dfdba23f2cd76722d9128b682e3c4c9fc7b
current exec_addr is 0x65015f08
Waiting for connection (30s)
wrong command or wrong mode detected, reboot your phone by pressing POWER and VOL_UP for 7-10 seconds.
```

TomKing062, 2024-07-20:
seems emmc type like pinned issue？

les1er1, 2024-07-20:
Yes, the unlock scripts for RMX3760 and RMX3762 use UFS memory type, but the phone has eMMC installed

TomKing062, 2024-07-21:
added a spl-patcher (by hexpatch) for new type (vab type) splloader, will rewrite batch and upload later

commit 98535676 (2024-07-21): (fix #100) add splloader patcher
closed 2024-07-21 by TomKing062