TomKing062/CVE-2022-38694_unlock_bootloader #101: Support for Umidigi Active T1

Arkbird1000, 2024-07-26:
I have been trying to follow the add device steps for my Umidigi Active T1 (T616 SoC) but am having difficulty in getting anywhere. This is probably because I am very unfamiliar with most of the parts of this process. I do have IDA but have not used it before trying to follow this tutorial.

I think I have managed to successfully extract the necessary SPL and FDL files from the .pac Umidigi provides. I have attached them in case my issues start with them and not something I am misunderstanding elsewhere.

[Active_T1.zip](https://github.com/user-attachments/files/16386851/Active_T1.zip)

Any help would be greatly appreciated.

TomKing062, 2024-07-28:
[ums9230_Umidigi_Active_T1.zip](https://github.com/user-attachments/files/16401982/ums9230_Umidigi_Active_T1.zip)

Arkbird1000, 2024-07-28:
Can confirm that worked. Thank you so much!

closed 2024-07-28 by Arkbird1000