TomKing062/CVE-2022-38694_unlock_bootloader #102: Still keep struggling to exit download mode A7cc

nirritia, 2024-08-01:
Hello, I entered download mode by this command
`adb reboot autodloader`. I used v1.5 and couldnt exit download mode.
issues #47 and #50 got me nothing.
Now im at 1.71 and still no go.
Could you please kindly guide me how to resolve the issue.
`C:\Users\nirritia\Downloads\38691_ud710_hisense_a7ccLast>spd_dump fdl fdl1-dl.bi
n 0x5500 fdl uboot-mod.bin 0x9efffe00 exec r splloader r teecfg r trustos r sml
e splloader e splloader_bak reset
branch:stable, sha1:47055dfdba23f2cd76722d9128b682e3c4c9fc7b
Waiting for connection (30s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND fdl1-dl.bin to 0x5500
EXEC FDL1
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
KEEP_CHARGE FDL1
SEND uboot-mod.bin to 0x9efffe00
FDL2: incompatible partition
EXEC FDL2
DISABLE_TRANSCODE
Reading Partition List
[=====                                   ] 12.5%
[==========                              ] 25.0%
[===============                         ] 37.5%
[====================                    ] 50.0%
[=========================               ] 62.5%
[==============================          ] 75.0%
[===================================     ] 87.5%
[========================================] 100.0%
dump_partition: user_partition+0x0, target: 0x8000, read: 0x8000
  0                            splloader 256KB
  1                             miscdata 5MB
  2                                 misc 1MB
  3                            nr_fixnv1 8MB
  4                            nr_fixnv2 8MB
  5                               prodnv 10MB
  6                        nr_runtimenv1 10MB
  7                        nr_runtimenv2 10MB
  8                             recovery 40MB
  9                              trustos 6MB
 10                          trustos_bak 6MB
 11                                  sml 1MB
 12                              sml_bak 1MB
 13                                uboot 1MB
 14                            uboot_bak 1MB
 15                            uboot_log 4MB
 16                                 logo 7MB
 17                            fbootlogo 7MB
 18                              l_pmsys 1MB
 19                              l_agdsp 6MB
 20                            gnssmodem 1MB
 21                             wcnmodem 10MB
 22                              persist 2MB
 23                               nr_spl 1MB
 24                               nr_sml 1MB
 25                             nr_uboot 1MB
 26                              nr_boot 35MB
 27                             nr_pmsys 1MB
 28                             nr_agdsp 6MB
 29                             nr_modem 40MB
 30                             nr_v3phy 8MB
 31                             nr_nrphy 8MB
 32                            nr_nrdsp1 5MB
 33                            nr_nrdsp2 5MB
 34                           nr_deltanv 2MB
 35                               teecfg 1MB
 36                           teecfg_bak 1MB
[... 170 lines cut ...]
[======================                  ] 55.1%
[=======================                 ] 57.8%
[========================                ] 60.2%
[=========================               ] 62.5%
[==========================              ] 65.2%
[===========================             ] 67.6%
[============================            ] 70.3%
[=============================           ] 72.7%
[==============================          ] 75.0%
[===============================         ] 77.7%
[================================        ] 80.1%
[=================================       ] 82.8%
[==================================      ] 85.2%
[===================================     ] 87.5%
[====================================    ] 90.2%
[=====================================   ] 92.6%
[======================================  ] 95.3%
[======================================= ] 97.7%
[========================================] 100.0%
dump_partition: sml+0x0, target: 0x100000, read: 0x100000
Answer "yes" to confirm the "erase partition" command:`
[image]
[image]
[image]
Thank you.

TomKing062, 2024-08-01:
spd_dump fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec w splloader u-boot-spl-16k-sign.bin reset

nirritia, 2024-08-01:
[image]
still no response

closed 2024-08-15 by TomKing062

lpjs31, 2024-08-22:
I'm having the same issue, stuck in download mode. I've been running the unlock batch but no response, even when using 
> spd_dump fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec w splloader u-boot-spl-16k-sign.bin reset
Any update on a way out of this? Thanks

TomKing062, 2024-08-23:
hisense provide cn rom in chinese forum(hard to download even in china)

the script only modified splloader, since recover spl not help, maybe uboot also need to be flashed. 
however i didn't save original uboot myself
see if there is fdl2-sign.bin in the folder(old release may have it, i deleted that in last unloading)
`spd_dump fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec w uboot fdl2-sign.bin w splloader u-boot-spl-16k-sign.bin reset`
or
`spd_dump fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec w uboot uboot-sign.bin w splloader u-boot-spl-16k-sign.bin reset`

lpjs31, 2024-08-23:
Hey TomKing,

Thank you so much for your help.

Here is what I have in the folder right now, no fdl2-sign.bin

[image: image.png]

Running the commands anyway gets me fopen(load) failed on both commands.

KrisHaz, 2024-09-10:
I am stuck on this as well. Not sure what to do, phone is frozen.