TomKing062/CVE-2022-38694_unlock_bootloader #103: Doubt

ZakitheBig, 2024-08-01:
Is there any universal solution for T310?

Thanks

p3y0t3, 2024-08-01:
What have you tried so far, regarding T310? I had zero luck, mainly because
of not knowing new sign key.

ZakitheBig, 2024-08-02:
> What have you tried so far, regarding T310? I had zero luck, mainly because of not knowing new sign key.
> […](#)

Well, I managed to partially unlock it through a website I found on the github pages. I can use the gsi normally, but the root doesn't work even after flashing the vbmeta-sign, so I looked for some tutorials about it and signed my boot.img (already with magisk) with public keys, from hovatek, to my surprise it worked, but every time magisk updates I have to do this process which is quite annoying.
So I was wondering if there was a universal method for the t310 and avoid all this work.

page: https://unisoc-android.github.io/subut/
hovatek: https://www.hovatek.com/forum/thread-32664.html

TomKing062, 2024-08-15:
t310 doesn't have universal unlock yet
---
hovatek guide is TOTALLY OUTDATED
use stock vbmeta and only sign boot with https://github.com/TomKing062/vendor_sprd_proprietories-source_packimage/blob/main/sign_image/config-unisoc/rsa4096_boot.pem

closed 2024-08-15 by TomKing062