TomKing062/CVE-2022-38694_unlock_bootloader #105: Support for noname t310 phone

mjkillly, 2024-08-06:
Hello,I have an iphone-like phone from TEMU.It runs Android 11 and the chip is Unisoc T310.I have asked the seller to get a copy of pac firmware and extract necessary files.Would you please help me to unlock?Thanks.

[Noname_t310_android11.zip](https://github.com/user-attachments/files/16514362/Noname_t310_android11.zip)

TomKing062, 2024-08-17:
these are signed with unisoc keys
use identifier token method (unless your device can not enter bootloader or doesn't have vloume keys)

mjkillly, 2024-08-21:
> these are signed with unisoc keys use identifier token method (unless your device can not enter bootloader or doesn't have vloume keys)

I've tried identifier token method,but it shows "FAILED (remote: Unlock bootloader fail.)"

```
C:\Users\123\Downloads\unlockbl\adb-fastboot>fastboot oem get_identifier_token

(bootloader) Identifier token:

(bootloader) 323438295749584230685496323434

OKAY [  0.021s]
Finished. Total time: 0.021s

C:\Users\123\Downloads\unlockbl\adb-fastboot>fastboot flashing unlock_bootload
er signature.bin
Downloading 'unlock_message'
OKAY [  0.003s]
unlocking bootloader
FAILED (remote: Unlock bootloader fail.)
Finished. Total time: 0.111s
```

TomKing062, 2024-08-28:
[310.zip](https://github.com/user-attachments/files/16787252/310.zip)

closed 2024-08-28 by TomKing062

pidanxiangjiao, 2025-02-05:
> [310.zip](https://github.com/user-attachments/files/16787252/310.zip)

Hi , i have a t310 device coolpad bos-a0, and now  i try to  execute "fastboot oem get_identifier_token" but error occurs, no valid token print , is there any command or solution i can try? Thanks.