TomKing062/CVE-2022-38694_unlock_bootloader #106: Support for ZTE Nubia Neo 2 T820

Random-629671, 2024-08-07:
I tried both `ums9620_test_dramtype2` and `ums9620_ZTE_universal` but both return `usb_send_failed(0/1)` at `spd_dump exec_addr 0x65012f48 fdl spl-unlock.bin 0x65000800`
Other function (read_part, write_part) work as expected
I don't know which things you need to support this so i copy the whole folder up here (without boot_b because they are the same):
[ums9620_ZTE_universal.zip](https://github.com/user-attachments/files/16519162/ums9620_ZTE_universal.zip)

Or is there just a way to root this things? I also tried to sign boot and vbmeta and send it to the phone. The logo appear without `invalid OS` caution but still not able to boot.

TomKing062, 2024-08-07:
`spd_dump exec_addr 0x65012f48 fdl spl-unlock.bin 0x65000800` will always fail or timeout
the real unlock status is miscdata 0x2000-0x2040
if 0x2000-0x2040 is not empty but system say locked,then it's ZTE block the device

Random-629671, 2024-08-08:
I guess they blocked it...
Waiting for a way to root this thing :((

TomKing062, 2024-08-08:
[image]

Random-629671, 2024-08-08:
Happy to see that you are still working with this

TomKing062, 2024-08-09:
how can i bypass a factory signature auth

closed 2024-08-09 by TomKing062

p3y0t3, 2024-08-09:
> ID: <TomKing062/CVE-2022-38694_unlock_bootloader/issues/106/2277541194@
> github.com>

He has instructions in his github on how to patch fdl1/fdl2 to bypass
factory signing. Ive petsonally only done it for ums312, however it should
be possible elsewhere

Random-629671, 2024-08-12:
i already got IDA pro but i don't know how to start, even the instruction it not clear enough for me
i guess i will wait for like... a year later so someone will have a way to, at least, root this things
thanks

p3y0t3, 2024-08-12:
Ida does have a bit of a learning curve to it. I personally used binary
ninja when i failed with ida - and actually managed to get usable results...

Random-629671, 2024-08-12:
i died with hex
thanks for ur app, i guess i need few month to start from zero with hex
damn, hex things always hard

Petrunak, 2025-02-19:
> i died with hex thanks for ur app, i guess i need few month to start from zero with hex damn, hex things always hard

Do you have firmware dump for this device? 
I research it too

Random-629671, 2025-02-19:
> > i died with hex thanks for ur app, i guess i need few month to start from zero with hex damn, hex things always hard
> 
> Do you have firmware dump for this device? I research it too

I used spd_dump to get the file
Also i gived up after just a week and now i forgot everything bruh