TomKing062/CVE-2022-38694_unlock_bootloader #113: Hisense A7 stuck in download mode?

Nexyf, 2024-09-14:
I ran 
`adb reboot autodloader`

and then unlock.bat. 
Here's the console log:
`H:\Downloads\Compressed\ud710_hisense_a7>spd_dump exec_addr 0x3f28 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec r splloader r teecfg r trustos r sml e splloader e splloader_bak reset
branch:stable, sha1:47055dfdba23f2cd76722d9128b682e3c4c9fc7b
current exec_addr is 0x3f28
Waiting for connection (30s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND fdl1-dl.bin to 0x5500
SEND custom_exec_no_verify_3f28.bin to 0x3f28
EXEC FDL1
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
KEEP_CHARGE FDL1
SEND uboot-mod.bin to 0x9efffe00
FDL2: incompatible partition
EXEC FDL2
DISABLE_TRANSCODE
Reading Partition List
[=====                                   ] 12.5%
[==========                              ] 25.0%
[===============                         ] 37.5%
[====================                    ] 50.0%
[=========================               ] 62.5%
[==============================          ] 75.0%
[===================================     ] 87.5%
[========================================] 100.0%
dump_partition: user_partition+0x0, target: 0x8000, read: 0x8000
  0                            splloader 256KB
  1                             miscdata 5MB
  2                                 misc 1MB
  3                            nr_fixnv1 8MB
  4                            nr_fixnv2 8MB
  5                               prodnv 10MB
  6                        nr_runtimenv1 10MB
  7                        nr_runtimenv2 10MB
  8                             recovery 40MB
  9                              trustos 6MB
 10                          trustos_bak 6MB
 11                                  sml 1MB
 12                              sml_bak 1MB
 13                                uboot 1MB
 14                            uboot_bak 1MB
 15                            uboot_log 4MB
 16                                 logo 7MB
 17                            fbootlogo 7MB
 18                              l_pmsys 1MB
 19                              l_agdsp 6MB
 20                            gnssmodem 1MB
 21                             wcnmodem 10MB
 22                              persist 2MB
 23                               nr_spl 1MB
 24                               nr_sml 1MB
 25                             nr_uboot 1MB
 26                              nr_boot 35MB
 27                             nr_pmsys 1MB
 28                             nr_agdsp 6MB
 29                             nr_modem 40MB
 30                             nr_v3phy 8MB
 31                             nr_nrphy 8MB
 32                            nr_nrdsp1 5MB
 33                            nr_nrdsp2 5MB
 34                           nr_deltanv 2MB
 35                               teecfg 1MB
 36                           teecfg_bak 1MB
[... 256 lines cut ...]
[2024-09-14 19:49:20:805] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:20:805] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
...
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:251] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:252] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:252] Read: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2024-09-14 19:49:35:252] Call Clear() fail, [ErrorCode:0x00000016].`

Nexyf, 2024-09-15:
I believe my stock firmware got erased before a backup was successful. Can anyone share stock bin files (splloader.bin etc.)

Nexyf, 2024-09-17:
Solved it. As I suspected, the splloader was deleted when I did adb reboot autodloader. Don't run that command unless you know what you are doing folks!

closed 2024-10-02 by TomKing062

VolaLuc, 2024-11-04:
> Solved it. As I suspected, the splloader was deleted when I did adb reboot autodloader. Don't run that command unless you know what you are doing folks!

Hey, stuck with my phone too. I don't know what to do. How did you solve th issue?