TomKing062/CVE-2022-38694_unlock_bootloader #116: JamboPhone (ums9230 (using universal unlock)) reboot itself after running ./spd_dump exec_addr 0x65015f08 fdl spl-unlock.bin 0x65000800

Bananduck, 2024-09-18:
My phone just rebooted itself after running that command, why though? here's the log:
```d3formed@debian:~/Downloads/ums9230_universal_unlock$ ./spd_dump exec_addr 0x65015f08 fdl spl-unlock.bin 0x65000800
branch:stable, sha1:09b8849b6a6c4738f5d6b843145fbfec1bba90db
current exec_addr is 0x65015f08
Waiting for connection (30s)
libusb_control_transfer ok
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND spl-unlock.bin to 0x65000800
SEND custom_exec_no_verify_65015f08.bin to 0x65015f08
EXEC FDL1
usb_recv failed : LIBUSB_ERROR_TIMEOUT
CHECK_BAUD FAIL
usb_recv failed : LIBUSB_ERROR_TIMEOUT
CHECK_BAUD FAIL
usb_recv failed : LIBUSB_ERROR_TIMEOUT
CHECK_BAUD FAIL
wrong command or wrong mode detected, reboot your phone by pressing POWER and VOL_UP for 7-10 seconds.
```

and also there is an unexpected response if it was a bad sign or something during running the `spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec w uboot fdl2-cboot.bin reset`:
[image]

**still unable to get unlocked :(**

PeratX, 2024-09-20:
can you please upload you backup bins? I got my phone bricked. The same model as yours.

PeratX, 2024-09-20:
@TomKing062 Any help to restore the original splloader? I've executed the last command and no luck.
```
.\spd_dump exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec r boot w splloader u-boot-spl-16k-sign.bin w uboot uboot_bak.bin  w misc misc-wipe.bin reset
branch:stable, sha1:b8d3cea198a3fa9f493632e9c0f60a6ae75339a8
current exec_addr is 0x65015f08
Waiting for connection (30s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND fdl1-dl.bin to 0x65000800
SEND custom_exec_no_verify_65015f08.bin to 0x65015f08
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
KEEP_CHARGE FDL1
SEND fdl2-dl.bin to 0x9efffe00
FDL2: incompatible partition
EXEC FDL2
DISABLE_TRANSCODE
Reading Partition List
[=====                                   ] 12.5%
[==========                              ] 25.0%
[===============                         ] 37.5%
[====================                    ] 50.0%
[=========================               ] 62.5%
[==============================          ] 75.0%
[===================================     ] 87.5%
[========================================] 100.0%
dump_partition: user_partition+0x0, target: 0x8000, read: 0x8000
  0                            splloader 256KB
  1                               prodnv 64MB
  2                             miscdata 1MB
  3                                 misc 1MB
  4                            trustos_a 6MB
  5                            trustos_b 6MB
  6                                sml_a 1MB
  7                                sml_b 1MB
  8                              uboot_a 3MB
  9                              uboot_b 3MB
 10                            uboot_log 16MB
 11                                 logo 8MB
 12                            fbootlogo 8MB
 13                           l_fixnv1_a 2MB
 14                           l_fixnv2_a 2MB
 15                           l_fixnv1_b 2MB
 16                           l_fixnv2_b 2MB
 17                         l_runtimenv1 2MB
 18                         l_runtimenv2 2MB
 19                              persist 2MB
 20                            l_modem_a 25MB
 21                            l_modem_b 25MB
 22                          l_deltanv_a 1MB
 23                          l_deltanv_b 1MB
 24                             l_gdsp_a 10MB
 25                             l_gdsp_b 10MB
 26                             l_ldsp_a 20MB
 27                             l_ldsp_b 20MB
 28                            l_agdsp_a 6MB
 29                            l_agdsp_b 6MB
 30                             pm_sys_a 1MB
 31                             pm_sys_b 1MB
 32                             teecfg_a 1MB
 33                             teecfg_b 1MB
 34                         hypervsior_a 10MB
 35                         hypervsior_b 10MB
 36                               boot_a 64MB
 37                               boot_b 64MB
 38                        vendor_boot_a 100MB
[... 87 lines cut ...]
partition_size_pc: uboot, 0x400000
file size : 0xfdf58
[==                                      ] 6.3%
[=====                                   ] 12.6%
[=======                                 ] 18.8%
[==========                              ] 25.1%
[============                            ] 31.4%
[===============                         ] 37.7%
[=================                       ] 43.9%
[====================                    ] 50.2%
[======================                  ] 56.5%
[=========================               ] 62.8%
[===========================             ] 69.0%
[==============================          ] 75.3%
[================================        ] 81.6%
[===================================     ] 87.9%
[=====================================   ] 94.1%
unexpected response (0x00a2)
unexpected response (0x0084)
Answer "y" to confirm the "write partition" command: y
partition_size_pc: misc, 0x100000
file size : 0x800
[========================================] 100.0%
load_partition: misc, target: 0x800, written: 0x800
```

PeratX, 2024-09-20:
I've fixed the problem by restoring splloader and uboot from the original phone. thx!
But I still cannot unlock the bootloader encounterting the same issue.

TomKing062, 2024-09-21:
spd_dump_0917 has a bug, will update soon

TomKing062, 2024-09-21:
new zip unloaded

TomKing062, 2024-10-18:
"spl-unlock.bin" is designed to reboot device

closed 2024-10-18 by TomKing062

PeratX, 2025-12-27:
Yes, but I left it on my computer at home. I might come home a month later. Then I'll try to find those files and upload them.