TomKing062/CVE-2022-38694_unlock_bootloader #117: doing ota update for rmx3231

Denzy7, 2024-09-21:
i have been rooted with unlocked bootloader for a while. when doing update, i get  stuck on downloading update and it fails. where is the problem here?

Denzy7, 2024-09-21:
I get this from dmesg
```
[155934.288842] fec: <715> error: parse_avb_image: failed to find avb hashtree descriptor
[155934.288859] fec: <715> error: fec_open: failed to parse avb image.
```
and update engine log
[update_engine.20240919-143613.txt](https://github.com/user-attachments/files/17083223/update_engine.20240919-143613.txt)

zoomver, 2024-09-21:
Hello Denzy7.
If the discussion is about updating the firmware of a smartphone:
Regardless of the type and manufacturer of the processor, updating the firmware over the air, via SD Card, through a firmware program, the smartphone must be without root access (stock boot.img, stock vbmeta.img).
Failure to comply with this rule leads to failures, failures, bootloop. On some smartphones, over-the-air updates require a locked bootloader.
Good luck to you.

closed 2024-09-24 by Denzy7