TomKing062/CVE-2022-38694_unlock_bootloader #118: unlock fail (device got removed itself and went back)

d3formed, 2024-09-22:
new zip unloaded

_Originally posted by @TomKing062 in https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/issues/116#issuecomment-2365016378_
            

```
PS C:\Users\Huy\Downloads\ums9230_universal_unlock> .\spd_dump.exe exec_addr 0x65015f08 fdl spl-unlock.bin 0x65000800
branch:stable, sha1:ab12243f174ba4a37f28feb5cc729b91331fcc83
current exec_addr is 0x65015f08
Waiting for connection (30s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND spl-unlock.bin to 0x65000800
SEND custom_exec_no_verify_65015f08.bin to 0x65015f08
EXEC FDL1
CHECK_BAUD FAIL
device removed, exiting...
PS C:\Users\Huy\Downloads\ums9230_universal_unlock>
```

UartComm_COM_6_2024_09_22.log:
```
[2024-09-22 14:11:51:620] ===============================C:\Users\Huy\Downloads\ums9230_universal_unlock\spd_dump.exe
[2024-09-22 14:11:54:439] Read: Call ClearCommError() fail, [ErrorCode: 0x000003E3].
[2024-09-22 14:11:54:951] Call Clear() fail, [ErrorCode:0x00000016].

```

d3formed, 2024-09-22:
*also FDL2 responded with "Incompatible partition(s)"  when running `spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec r splloader r uboot e splloader e splloader_bak reset`

TomKing062, 2024-09-22:
spl-unlock.bin is designed to a reboot
if it not reboot, that's a problem

TomKing062, 2024-09-22:
and i re uploaded again
to fix batch(change %errorlevel% to !errorlevel!)

d3formed, 2024-09-23:
> and i re uploaded again
> to fix batch(change %errorlevel% to !errorlevel!)

I think the phone bricked, i did restore everything but nothing working

TomKing062, 2024-09-23:
the unlock procedure only modify uboot and splloader

d3formed, 2024-09-23:
> the unlock procedure only modify uboot and splloader

I did all the steps, but it didn't reboot itself after that

TomKing062, 2024-10-18:
i fixed issue in batches about losing backup in second run, but honestly ,it's hard to unbrick when partition backup is already lost

your case, the issue is fdl2-cboot wrote to device and uboot backup lost due to error in previous batch

closed 2024-10-18 by TomKing062

d3formed, 2024-10-20:
> i fixed issue in batches about losing backup in second run, but honestly ,it's hard to unbrick when partition backup is already lost
> 
> your case, the issue is fdl2-cboot wrote to device and uboot backup lost due to error in previous batch

WAIT.. i think i have a few backup of uboot since the first tool runs, idk if it will work